revtel/react-native-nfc-manager #213: 2.0 beta.4 - Still errors with utf8 bytes to string

NicholasBertazzon, 2019-10-27:
Hi, it's always me, and I'm here again for the utf8 to string converter.
I've seen that you inserted another utf8 to string algorithm instead of the one I committed.

Problem is, chars like § doesn't show anymore :(

My commit:
([https://github.com/whitedogg13/react-native-nfc-manager/commit/8d6b08ec37316ac5c965f099003e88a473f5494b#diff-816850ddc2b2655e94482a43cb29a1ac](url)

whitedogg13, 2019-10-28:
Hi @NicholasBertazzon oops, sorry for that. Since I'm not a text encoding algorithm master, just pick one found in gist. Can you kindly give me a PR? I'll be glad to accept that.

NicholasBertazzon, 2019-10-28:
@whitedogg13 done, I copied my commit from Nov. 2018

whitedogg13, 2019-10-29:
@NicholasBertazzon thanks for the PR, this patch has been published in `2.0.1`

closed 2019-10-29 by whitedogg13