revtel/react-native-nfc-manager #215: NFCManager.registerTagEvent got 3 arguments expected 2

gosuisez, 2019-10-28:
I got the following error

[image]

whitedogg13, 2019-11-04:
@gosuisez Sorry for the late resopse. It looks like a version mismatch between library and your client code. Please show me your code and library version for further info.

bigescalade, 2019-11-15:
this is happening for me also on iOS 13, version 2.0.1.

`index 2 beyond bounds [0 .. 1]' was thrown while invoking registerTagEvent on target NfcManager with params (
    "",
        {
        invalidateAfterFirstRead = 0;
        isReaderModeEnabled = 0;
        readerModeFlags = 0;
    },
    1923
)`

bigescalade, 2019-11-20:
This was an issue for my project using codepush, it was reverting the library back to an older version.

closed 2020-02-26 by whitedogg13

anaraa3, 2023-02-14:
> This was an issue for my project using codepush, it was reverting the library back to an older version.

Hi @bigescalade,
My project using codepush, How to solve this problem when using codePush.

Hi @bigescalade,
My project using codepush, how to solve this problem when using codepush.

Please sharing your solution^

Thanks

victorcorreae, 2023-09-29:
Hey @anaraa3 
I've had updated some packages including

> react-native-nfc-manager

So It was a version problem, so what worked for me was to clear all data, do a fresh install and rebuild.