revtel/react-native-nfc-manager #216: iOS 13 not working

guoxianzhe, 2019-10-30:
environment:
    "react-native-nfc-manager": "^2.0.1",
    "react-native": "0.61.1",

when i want to read card tag on the iOS13,the screen always stay at this step:
[image]

xcode shows:
[image]

is that something wrong what i had?

whitedogg13, 2019-11-04:
@guoxianzhe Sorry for the late response. The UI is perfectly normal, it's just the default prompt shown by iOS, and it will dismiss once the tag is scanned.

guoxianzhe, 2019-11-06:
@whitedogg13 sorry,when the ic card is in contact with the iphone, the iphone has no reaction. 
the screen always stay at this step.

acasademont, 2019-11-07:
@whitedogg13 we're seeing the same, if we send the comand first we see the iOS prompt but nothing in return.

whitedogg13, 2019-11-07:
Hi all, here’re some updates. We have found that some tags cannot be correctly scanned by iOS 13, as discussed in #210 . Can you guys report what tags you’re using?

guoxianzhe, 2019-11-07:
i used 'example/AppV2.js' to get the tag
[image]

saw the iOS prompt but nothing in return.

acasademont, 2019-11-07:
@whitedogg13 sure, it's the NXP NTAG213

acasademont, 2019-11-07:
A normal NDEF scanning app like the ones you could do before iOS13 (GoToTags for example) works well with the tags and reads the NDEF record without problems.

whitedogg13, 2019-11-07:
Could you share a minimal project for me to debug？since I cannot reproduce this one, it’s hard to guess what’s going wrong.

acasademont, 2019-11-07:
@whitedogg13 ok we found it, seems like we enabled the capabilities on our Identifiers page on the apple developers website but we didn't actually add the NFC capability to our Xcode project...I believe @guoxianzhe might be facing the same issue. For reference, the .entitlements file should have these lines:

[image]

acasademont, 2019-11-07:
Sorry and thanks for your time!!

acasademont, 2019-11-07:
The problem is that we added the entitlements as the guide https://github.com/hansemannn/iOS-NFC-Example suggest, but there's a new entitlement that needs to be added. The correct entitlements file is 

```xml
<key>com.apple.developer.nfc.readersession.formats</key>
<array>
  <string>NDEF</string>
  <string>TAG</string>
</array>
```

the "TAG" one is the new one in iOS 13 that the guide doesn't mention. We didn't see until we added the entitlements via Xcode GUI

whitedogg13, 2019-11-08:
@acasademont Awesome! Sorry for not keeping the doc up to date, would you mind sending a PR to update readme? I will be very happy to accept that!

acasademont, 2019-11-08:
Sure, we'll try ;)

acasademont, 2019-11-08:
Done in #217 @guoxianzhe could you confirm if this also fixes your issue?

guoxianzhe, 2019-11-13:
@acasademont @whitedogg13 

I changed my entitlements files like this:
[image]

It seems still not working :( 

I tried to use example directory in my project, it's still not working on iOS13

about iOS setting:
i just finish these steps:
[image] info.plist

[image] .entitlements

[image] 

it seems all right of my setting

inkCrazy, 2019-11-14:
我也有相同的问题，我使用的版本是"react-native-nfc-manager": "^1.2.2",
桥接文件中使用的是NFCNDEFReaderSession类，don't have the class of 'NFCTagReaderSession'.  
NFCTagReaderSession allows to scan native tags like ISO14443, ISO15693, ISO18092
NFCNDEFReaderSession supports discovery of NDEF tag objects and read/write operations for those
NFCNDEFReaderSession无法识别没有支持的芯片，我使用的芯片类别是ISO 14443-3A NXP Mifare Ultralight-NTAG213

this is iOS13 demo: is good for NFCTagReaderSession
```
#import "ViewController.h"
#import <CoreNFC/CoreNFC.h>
#import <MessageUI/MessageUI.h>

@interface ViewController ()<NFCTagReaderSessionDelegate>

@property (weak, nonatomic) IBOutlet UILabel *readText;
@property(strong,nonatomic)NFCTagReaderSession *session;

@end

@implementation ViewController

- (void)viewDidLoad {
    [super viewDidLoad];
}

- (IBAction)scanNFC:(UIButton *)sender {
    self.session = [[NFCTagReaderSession alloc] initWithPollingOption:(NFCPollingISO14443 | NFCPollingISO15693 | NFCPollingISO15693) delegate:self queue:nil];
    self.session.alertMessage = @"读取卡片，请将卡片靠近手机";
    [self.session beginSession]; //开始识别 弹出识别提示框
}

- (void)tagReaderSession:(NFCTagReaderSession *)session didInvalidateWithError:(NSError *)error API_AVAILABLE(ios(13.0)) API_UNAVAILABLE(watchos, macos, tvos) {
    NSLog(@"didInvalidateWithError => %@", error.localizedRecoverySuggestion);
}

- (void)tagReaderSessionDidBecomeActive:(NFCTagReaderSession *)session API_AVAILABLE(ios(13.0)) API_UNAVAILABLE(watchos, macos, tvos) {
    NSLog(@"tagReaderSessionDidBecomeActive => %@", session);
}

-(void)tagReaderSession:(NFCTagReaderSession *)session didDetectTags:(NSArray<__kindof id<NFCTag>> *)tags{
    NSLog(@"didDetectTags => %@", tags);
}
```

whitedogg13, 2019-11-19:
@inkCrazy Please use v2.0.1 instead, v1 contains no iOS 13 support.

whitedogg13, 2019-11-19:
@guoxianzhe Your setting looks good to me. Can you confirm your tag can be scanned by other iOS NFC app like NXP tag info? Just to make sure it's not related to the tag.

guoxianzhe, 2019-11-27:
@whitedogg13 
Yes,the tag that I used is perfectly by other iOS NFC app.

mencomao, 2019-12-11:
On ios13, I scan a blank Ntag213 first, and then scan a Ntag213 with data. At this time, there is no response. If I scan the Ntag213 with data directly, the data can be read.

mencomao, 2019-12-11:
@whitedogg13 On ios13, I scan a blank Ntag213 first, and then scan a Ntag213 with data. At this time, there is no response. If I scan the Ntag213 with data directly, the data can be read.

vadermemo, 2020-01-22:
Same issue here

"dependencies": {
    "react": "16.9.0",
    "react-native": "0.61.4",
    "react-native-nfc-manager": "^2.0.1"
  },

Dont work on iOS 13 and iphone 11.

Even when I active the shorcuts on Iphone and make the configuration to Automatization> Personal>NFC my cards are readables correctly.

Any can resolve this bug!?

Best regards.

vadermemo, 2020-01-22:
> Hi all, here’re some updates. We have found that some tags cannot be correctly scanned by iOS 13, as discussed in #210 . Can you guys report what tags you’re using?

Same issue with clone cards Arduino.

vadermemo, 2020-01-22:
> > Hi all, here’re some updates. We have found that some tags cannot be correctly scanned by iOS 13, as discussed in #210 . Can you guys report what tags you’re using?
> 
> Same issue with clone cards Arduino.

Even doesn´t work with NFC TAG Keychain 13.56MHz for Arduino (Blue)

dmorais92, 2020-01-27:
Followed the Mifare example and cannot scan Mifare Classic or Plus and it doesn't seem to be working on iOS 13, iPhone X

Megatron4537, 2020-02-05:
Heads up for those struggling with NTAG213, my NTAG213 supports MifareUltralight, NfcA and Ndef, on iOS, using the pair [setEventListener,registerTagEvent] I can detect the tags but not read its UID, to do so i had to use the mifare example and use [requestTechnology(NfcTech.MifareIOS), getTag] (this gives the tag UID)

whitedogg13, 2020-02-06:
Hi @Megatron4537 thanks for sharing! 

What you mentioned is indeed the correct solution. BTW, the limitation around `uid` is because iOS won’t give us the `uid` when requesting `Ndef` technology, and it’s is a platform limitation we cannot bypass.

Thanks!

whitedogg13, 2020-02-07:
Hi all, some update for people facing the issue `only see the iOS prompt but nothing can be scanned`:

Please try to use [this example](https://github.com/whitedogg13/nfc-test-app) to see if it works.
(in issue #250, some developer reports the issue been fixed by trying above example, although the root cause is not identified)

vanronaldo2, 2020-06-12:
Hello.
I have the same issue, it fixed by following it.
https://github.com/whitedogg13/react-native-nfc-manager/issues/289#issue-616577045

bingbing720, 2020-06-12:
I have the same issue, it fixed by following the below.
https://github.com/whitedogg13/react-native-nfc-manager/issues/289

guoxianzhe, 2020-10-23:
I fixed it by use other card

closed 2020-10-23 by guoxianzhe