revtel/react-native-nfc-manager #218: how to use v2 with RN0.5x?

Symous, 2019-11-09:
Hi, friend, years ago, I used this lib v0.x to build my app. now I noticed that you have brought about lots of new features on iOS 13 with v2. I have a few questions:
1. how can I use this lib v2 with RN0.5x which not support auto link?
2.Is now possible to obtain the NFC tag id with this lib v2 on iOS just like android?
3.If the answer of Q2 is yes, whether there is any limitation of the preview of App Store? 

Thanks.

whitedogg13, 2019-11-10:
Hi @Symous Thanks for your support all the time! 

Regarding your questions:
1. v2 should works well with RN 0.59.10, but I'm not sure about previous versions. The reason is because we need XCode 11 to access new NFC feature on iOS 13, and RN before 0.59.10 seems to have issues when works with XCode 11 (0.59.10 has some back-ported features from 0.60, so it's fine), some info [here](https://github.com/facebook/react-native/issues/25181).
2. yes, we should be able to obtain uid for most NFC tags on iOS 13, the only exception is `Felica`.
3. I think no, since we obtain the id from public iOS API, so there shouldn't be any limitation.

BTW, the API has also changed from V1 to V2, please see example directory, the file has been postfixed according to its supporting version.

Just tell me if there's still anyting unclear!

Symous, 2019-11-11:
Thanks for your reply. I have built a new app with to test the iOS13 NFC features: 
react-native: 0.61.4
react-native-nfc-manager: 2.0.1
XCode:11

I still have a few questions: 
1. I found type of Mifare Classic NFC tag can not be read(nothing happened when approach to iPhone) while Mifare Desfire works(get uid). I want to know that whether it is disabled by Apple to reading Mifare Classic tags or there is any other extra configurations I missed.
2. This lib v2.x(feature to get uid on both android and iOS) can work with RN0.59.10 as long as I upgrade XCode to V11 right?

Thanks again.

whitedogg13, 2019-11-12:
Hi @Symous 

1. Regarding Mifare Classic card, I don't think you have missed anything, since I cannot read them either during some preliminary test. I will do more tests and keep you posted.
2. This library should work well in 0.59.10 with manually `react-native link`!

Thanks!

closed 2021-05-11 by Symous