revtel/react-native-nfc-manager #219: requestTechnology Ndef doesn't ever get resolved

AnthonyGolo, 2019-11-14:
Greetings, 
there is an issue concerning requestTechnology method. It does neither get resolved nor rejected forever. 
Omitting it is not permittable, there is an exception 'no tech request available' raised in that case.

Following snippet is a part of my NfcAndroid service class:

`
...
writeNfcTag = async () => {
    try {

      let resp = await NfcManager.requestTechnology(NfcTech.Ndef);
      // isn't ever reached
      let ndef = await NfcManager.getNdefMessage();
      let payload = buildTextPayload(this.textToSend);
      let bytes = Ndef.encodeMessage([buildRecordFromText(payload)]);
      await NfcManager.writeNdefMessage(bytes);
      console.log('!SUCCESSFULLY WROTE NFC TEXT');
    } catch (ex) {
      console.warn('!EXCEPTION', ex);
    }
  }
...
`

whitedogg13, 2019-11-19:
Hi @AnthonyGolo sorry for the late response. The promise you mentioned only get resolved when a tag is detected, or via an explicitly `cancelTechnology` call. Once it's resolved, you have an active session connects to the tag and so you can do further actions like NDEF writing.

LordHydra1, 2020-01-16:
Every time i turn on the app and try to use the nfc with 2 different phone the write method it always stops at the first try{requestTecnology(bla,bla,bla)} so i cant send anything and when i try to read i recieve a default tag of android who send me a link to a play store page @whitedogg13

taibui-vn, 2020-05-06:
"The promise you mentioned only get resolved when a tag is detected" 
Do you mean on NfcEvents.DiscoverTag event, it get resolved? In my case, when requestTechnology is not resolved, this event is not fired.

closed 2021-05-26 by github-actions[bot]