revtel/react-native-nfc-manager #221: Iphone 7  IO 13+ Isodep does not see ADPU card

orhanoksuzz, 2019-11-25:
I use code example
https://github.com/whitedogg13/react-native-nfc-manager/blob/master/example/AppV2Apdu.js

It works fine on Android but does not see the card on the iphone side.

https://github.com/whitedogg13/react-native-nfc-manager/blob/master/example/AppV2Mifare.js

In this example, I can see the same card information, but I cannot send an ADPU command.

thecalvinchan, 2019-11-26:
This fixed it for me https://github.com/whitedogg13/react-native-nfc-manager/issues/216#issuecomment-551168459

orhanoksuzz, 2019-11-26:
> This fixed it for me [#216 (comment)](https://github.com/whitedogg13/react-native-nfc-manager/issues/216#issuecomment-551168459)

The solution failed for me.I can read tags with miframe, but when I select Isodep format to send adpu command, the card does not detect

orhanoksuzz, 2019-11-26:
`  _test = async () => {
    try {
      let tech = NfcTech.IsoDep;
  
      let resp = await NfcManager.requestTechnology(tech, {
        alertMessage: 'Ready to send some APDU',
      });
     console.warn(resp)

      // the NFC uid can be found in tag.id
      let tag = await NfcManager.getTag();
      console.warn(tag)
   
      if (Platform.OS === 'ios') {
        resp = await NfcManager.sendCommandAPDUIOS(customapdu);
        } else {
        resp = await NfcManager.transceive(customApdu);
        }
      
     console.warn(resp)
      this._cleanUp();
    } catch (ex) {
       console.warn('ex',ex)
      this._cleanUp();
    }
  };`
Apple real device result:
error :" No tech matches"

puneetmahali, 2019-12-30:
I am also facing the same issue on iPhone 7 real Device - "No tech matches" when i am trying to read the e-Passport chips. Has anyone solved this problem?

whitedogg13, 2019-12-30:
Hi all, I can successfully send APDU to ePassport chip using iPhone 11 and this library. I will try to get a iPhone 7 and test it out.

For the testing, I simply use this [example](https://github.com/whitedogg13/react-native-nfc-manager/blob/master/example/AppV2Apdu.js).

Some extra configs are required. You should declare the AID in your plist, to read ePassport:
```
	<key>com.apple.developer.nfc.readersession.iso7816.select-identifiers</key>
	<array>
		<string>A0000002471001</string>
	</array>
```

You also need to add following into your entitlement:
```
<dict>
	<key>com.apple.developer.nfc.readersession.formats</key>
	<array>
		<string>NDEF</string>
		<string>TAG</string>
	</array>
</dict>
```

Also, please remember to enable NFC reading capability for your app in developer website: 
[image]

orhanoksuzz, 2019-12-31:
> Hi all, I can successfully send APDU to ePassport chip using iPhone 11 and this library. I will try to get a iPhone 7 and test it out.
>
> For the testing, I simply use this [...]

The cards I use do not have an ID. Guess the library has been updated. I haven't had a chance to try yet.I'll try it after Christmas.

closed 2021-05-26 by github-actions[bot]