revtel/react-native-nfc-manager #225: Writing on NdefFormatable does not work

jamesjypark, 2019-11-29:
Hello!

I am having issues with writing data on an NdefFormatable object which fails and invokes "fail to apply ndef tech" warning. 

This issue is caused by line 1174 in `android/src/main/java/community/revteltech/nfc/NfcManager.java`:
```
try {
  Log.d(LOG_TAG, "ready to writeNdef");
  Ndef ndef = Ndef.get(tag);
  if (ndef == null) {
    callback.invoke("fail to apply ndef tech"); // line 1174
  } else if (!ndef.isWritable()) { ...
```

Instead of invoking error if `ndef` is `null`, I think we can follow Android developers documentation and use NdefFormatable.[format](https://developer.android.com/reference/android/nfc/tech/NdefFormatable.html#format(android.nfc.NdefMessage)) to convert NdefFormatable to Ndef and then proceed to write on the NFC.

I am not too familiar with how to make changes to the repository and testing it so any help will be greatly appreciated :)

Thanks!
James

lucasbarbieri, 2021-05-05:
You find fix for this?

jamesjypark, 2021-05-05:
The way I remember solving this issue was by giving sufficient physical contact time (~3 seconds) between the NFC device and the tag for the data transmission to complete. 
The error was invoked when the NFC device and the tag were in contact for too short.

closed 2021-07-20 by github-actions[bot]

paichato, 2022-02-04:
It seems like they have updated the lib and it works well now on the version @3.13.0.