revtel/react-native-nfc-manager #226: Unable to get data to another device ?

ektasahu027, 2019-12-02:
Could you please share the process to get data from one device to another device.

whitedogg13, 2019-12-26:
@ektasahu027 sorry for the late respone. I will update the example once I'm available~

LordHydra1, 2020-01-16:
@whitedogg13 can you please try one app that read and one that write and try it on two different phone because i tryed both of your example I copy and pasted it and i can read the tag but i cant read what i want to send. I'm trying to send a simple string and i always get inside the tag method an array of bytes but not right one...could you please help me ?

LordHydra1, 2020-02-12:
Up

egislook, 2020-02-27:
> @whitedogg13 can you please try one app that read and one that write and try it on two different phone because i tryed both of your example I copy and pasted it and i can read the tag but i cant [...]

you just have to decode the array

closed 2021-05-26 by github-actions[bot]