revtel/react-native-nfc-manager #229: Share data between Android / IOS

rodgomesc, 2019-12-16:
It's possible share data between Android and IOS using this library?

whitedogg13, 2019-12-26:
Hi @rodgomesc Sorry for the late response. It's not possible to share data between iOS / Android using this library now. That's because this library only focus on "reader" mode of NFC, to communicate between devices, one of them should be a tag.

closed 2019-12-26 by whitedogg13

ifeoluwak, 2019-12-27:
I would like to understand something, can we share data between Android to Android or iOS to iOS?

rodgomesc, 2019-12-27:
@ifeoluwak Android <--> IOS using NFC

whitedogg13, 2019-12-28:
@ifeoluwak Currently it's not possible to perform iOS to iOS communication. For Android to Android communication can be accomplished with 'Android Beam' feature, which is `NfcManager.setNdefPushMessage` API.

amitbravo, 2021-03-24:
what are possibilities today ? can we share data between 2 devices ??

whitedogg13, 2021-03-24:
Hi @amitbravo No, those APIs are deprecated by Google~