revtel/react-native-nfc-manager #230: Javascript iOS support for ISO 15693

JimTeva, 2019-12-23:
Looks like @whitedogg13  did the work to integrate latest support in iOS 13 for tags ISO 15693 (and FeliCa) in ios/NfcManager.m
``` js
[NSNumber numberWithInt: NFCTagTypeFeliCa]: @"felica",
[NSNumber numberWithInt: NFCTagTypeISO15693]: @"iso15693",
```
But those two new types of tag are missing on the javascript side in NfcManager.js
```js
const NfcTech = {
  Ndef: 'Ndef',
  NfcA: 'NfcA',
  NfcB: 'NfcB',
  NfcF: 'NfcF',
  NfcV: 'NfcV',
  IsoDep: 'IsoDep',
  MifareClassic: 'MifareClassic',
  MifareUltralight: 'MifareUltralight',
  MifareIOS: 'mifare',
}
```
Looks like the work on FeliCa is not yet finished but the work on iso15693 looks done, so @whitedogg13 could you add it to NfcTech, please?

whitedogg13, 2019-12-26:
Hi @JimTeva Sorry for the late response. We haven't supported iso15693 yet, once it has more concrete progress I will update it here, thanks!

puneetmahali, 2019-12-30:
Hi @whitedogg13 ,
I am trying to read the e-Passport using react-native-nfc-manager with React -Native. Also, I am new in react-native(Javascript). When I run on iPhone device then I got it the error "No tech matches" from 
`(void)tagReaderSession:(NFCTagReaderSession *)session didDetectTags:(NSArray<__kindof id<NFCTag>> *)tags
{}`
method. It's detect the tag but not matching.
After that I need to write the code to read the passport data in Xcode natively or in react-native.

Please guide me little more here. Thanks in advance!

whitedogg13, 2019-12-30:
Hi @pheromonez for ePassport, you might need to use `IsoDep` NFC tech. Some extra setups are required to do this, as my comment in #221 .

Frans-L, 2020-01-04:
Hey @whitedogg13!  Do you have any estimation when NfcV (ISO 15693) support for iOS might be ready to test/use?

whitedogg13, 2020-01-05:
Hi @Frans-L , I plan to release a beta version next week, and I will definitely need your feedback and testing!

whitedogg13, 2020-01-08:
Hi all, I just published `2.1.0-beta.0`, which contains most ISO15693 support for iOS 13. 

Most of the core NFC APIs are implemented except for the multiple blocks related ones. Here's the supported API in this beta release:
```typescript
interface Iso15693HandlerIOS {
    getSystemInfo: (
      requestFloags: number,
    ) => Promise<{ dsfid: number, afi: number, blockSize: number, blockCount: number, icReference: number}>;
    readSingleBlock: (params: {flags: number, blockNumber: number}) => Promise<number[]>;
    writeSingleBlock: (params: {flags: number, blockNumber: number, dataBlock: number[]}) => Promise<void>; 
    lockBlock: (params: {flags: number, blockNumber: number}) => Promise<void>; 
    writeAFI: (params: {flags: number, afi: number}) => Promise<void>; 
    lockAFI: (params: {flags: number}) => Promise<void>; 
    writeDSFID: (params: {flags: number, dsfid: number}) => Promise<void>; 
    lockDSFID: (params: {flags: number}) => Promise<void>; 
    resetToReady: (params: {flags: number}) => Promise<void>; 
    select: (params: {flags: number}) => Promise<void>; 
    stayQuite: () => Promise<void>; 
    customCommand: (params: {flags: number, customCommandCode: number, customRequestParameters: number[]}) => Promise<number[]>;
    extendedReadSingleBlock: (params: {flags: number, blockNumber: number}) => Promise<number[]>;
    extendedWriteSingleBlock: (params: {flags: number, blockNumber: number, dataBlock: number[]}) => Promise<void>; 
    extendedLockBlock: (params: {flags: number, blockNumber: number}) => Promise<void>; 
  }
```

For example, please see [here](https://github.com/whitedogg13/react-native-nfc-manager/blob/ios-15693/example/AppV2Iso15693.js).

Since I only test a small part of the API for this beta release, please feel free to contribute or fire issues, and the codes are in branch `ios-15693`.

Thanks!

JimTeva, 2020-01-08:
Thank you so much @whitedogg13 Richie!!

closed 2020-01-17 by JimTeva

raydenvoldeskine, 2022-01-10:
Hi, is there a way I can use raw transceive() commands for iso 15963 tag for iOS? The issue is that I use NfcManager.transceive() for Android and it seem to work perfectly for all types of tags. GIven that; I already construct commands by myself. And, obviously, since we are in the cross platform environment per definition, I do not want to construct commands differently for iOS so I only need transceive(). Can you help with that?

raydenvoldeskine, 2022-01-10:
I tried to use custom command API on ISO 15693 ISO implementation and some commands worked however commands with .address implementation did not. Found only one reference that this is not supported: 

https://developer.apple.com/forums/thread/129242

Now starting from iOS 14 there is a different method which looks promising. Is it a lot of efforts to extend JS iOS implementation to get access to this one? 

https://developer.apple.com/documentation/corenfc/nfciso15693tag/3551933-sendrequestwithflag?language=objc

whitedogg13, 2022-01-10:
@raydenvoldeskine thanks for the update. From the API signature, it looks pretty like the Android's `.transceive` call, we will take a look at this soon.

raydenvoldeskine, 2022-01-11:
Thanks. I can offer a test possibility - currently I have a specific tag and specifc command with address mode which works on android.  I have never modified a react native component - would it work if I would just try to add the method locally (directly in the npm synced version) - similar to other methods?