revtel/react-native-nfc-manager #236: requestTechnology with multiple tech (2.1.0-beta)

jberns88, 2020-01-15:
First off, thanks for your hard work with the ISO15693 stuff for IOS. I've managed to get everything working that I need. I've just got the one issue below.

On iOS I'm trying to detect Ndef and ISO15693 at the same time but when Ndef is within the requestTechnology list ISO15693 won't detect.

I can see that there is an if statement around `hasNdefTech` that does stuff with registerTagEvent that I don't really understand and was basically wondering if this is because it's not possible to detect both techs at the same time or is something else going on?

When I only put ISO15693 in the tech list It does still scan the Ndef card but just says "not in tech list" so this leaves me to believe it is possible to detect both?

jberns88, 2020-01-16:
A couple other problems I've found with 2.1.0-beta:

1, On iOS getTag doesn't work with Ndef. I think I understand why but maybe we make a wrapper to return something like what android returns?

2, getNdefMessage() returns an object containing ndefMessage on Android but on iOS it returns just the message.
```
//Android
NfcManager.getNdefMessage().then(({ndefMessage}) => ... )
//iOS
NfcManager.getNdefMessage().then((ndefMessage) => ... );
```
3, Cancelling on iOS doesn't resolve my promise. I know there is `setEventListener` for session closed but it would be nice if it rejected my `requestTechnology` promise.

JimTeva, 2020-01-17:
I have the same requirement: need to be able to read different tags with the same requestTech (both NfcTech.MifareIOS and NfcTech.Iso15693IOS)
It looks like it is not possible... 
@whitedogg13  any idea?

whitedogg13, 2020-01-18:
Hi @jberns88 Thanks for those good catches! 

The issues 2/3 have been fixed in `v2.1.0-beta.3`. For issue 1, I use a different approachby enhance the `getTag` API for iOS, which contains `NdefMessage` for the discovered tag regardless of the requested technology! 

So you can get the `NdefMessage` without requesting for `Ndef` tech like this:
```javascript
await NfcManager.requestTechnology(NfcTech.MifareIOS);
let tag = await NfcManager.getTag(); 
console.log(tag.ndefMessage); // <-- now you have `tag.ndefMessage`!
```

@JimTeva You should be able to register multiple techs by passing array of techs into `requestTechnology`. When a new tag is discovered, the library will try to connect to the tag via your requested tech in order.

However in current implementation we favor `Ndef` tech over others, so if you use, for example, `Ndef` and `Mifare` at the same time, the `Mifare` technology will not take effect. If what you want is to read `NdefMessage` and perform some technology specific task at the same time, you don't need to register for `Ndef` and can simply use the `getTag` method to retrieve current `NdefMessage` in the tag. This feature is implemeted in `v2.1.0-beta.3`.

Hope that helps, and please tell me if there're still any issues!

Thanks for you guys!

jberns88, 2020-01-18:
Thanks @whitedogg13 but I think we have a misunderstanding with #1. Simply getTag() throws and error on iOS "not even registered" but works ok Android when using a Ndef tag.

That's a killer with the Ndef favoured. I really need to be able to detect both as I don't know what the user is trying to scan. Could be Ndef or ISO15693. Is there anything we can do to fix this?  I'd love to help but I've never written a line of swift in my life :-(. The Android implemention works perfectly letting detect Ndef or ISO15693.

whitedogg13, 2020-01-21:
@jberns88 Got it. In the case you described, I think this is the closest approach:
```javascript
// first, register for ALL techs
const resolvedTech = await NfcManager.requestTechnology(
  [NfcTech.MifareIOS, NfcTech.Iso15693IOS, NfcTech.IsoDep], 
  { alertMessage: 'Ready to scan tags' }
);

// second, read the ndef,
// the tag resolved will contain NdefMessage regardless of the resolvedTech
const tag = await NfcManager.getTag(); 

// third, check the resolvedTech and perform specific task
if (resolvedTech === NfcTech.Iso15693IOS) {
  // do Iso15693 related stuff!
}
```

Some more details:
In iOS, the general `Ndef` type is tied to different delegate than the specific type like `Iso15693`, so confusion arised when registering `Ndef` and other tech types together. However, tag objects (including `NFCMifareTag`, `NFCISO7816Tag`, `NFCISO15693Tag`, `NFCFeliCaTag`) are all inherated from `NFCNDEFTag`, so we can always read `NdefMessage` from them.

Thanks!

jberns88, 2020-01-22:
@whitedogg13 I get what you're saying but I think there we aren't quite on the same wavelength.

**Ndef + Iso15693IOS** 
I have two different types of tags that I need to detect. One is an Ndef only tag NTAG213 and the other is an Iso15693/NfcV

When trying to detect the NTAG213 (ISO14443). This tag only contains Ndef and therefore **is** an ```NFCNDEFTag``` and no other kind of tech so the only way for me to discover it is by doing ```[NfcTech.Ndef]``` as it isn't part of any other standard. 

So when doing  ```[NfcTech.Iso15693IOS]``` I cannot detect my NTAG213 as I get ```"No tech matches"```. I've even tried catching that error and calling ```getTag()``` anyway but then I get ```"Not connected".```

And as before when doing ```[NfcTech.Ndef, Nfc.Iso15693IOS]``` I cannot detect the Iso15693IOS due to reasons you explained. Though this tech list is ultimately what I would like to achieve. 

**Cancel not rejecting promise**
I've just tried the update you made and I think you fixed a bug that wasn't the one I was talking about haha. When pressing the native cancelled button that appears on the screen my promise isn't rejected. As a work around I did the following:

```
const wrapInCancelled = (callback) => {
    const removeListener = () => NfcManager.setEventListener(NfcEvents.SessionClosed, null);

    return new Promise((resolve, reject) => {

        const onCancelled = () => {
            reject("cancelled");
            removeListener();
        }

        NfcManager.setEventListener(NfcEvents.SessionClosed, onCancelled);

        callback()
            .catch(e => {
                removeListener();
                throw e;
            })
            .then(resolve)
            .catch(reject);

    });
}
....
wrapInCancelled(() => NfcManager.requestTechnology([NfcTech.Ndef]))
    .then(...)
```

Sorry to be a pain! I'd love to help you but I've no idea what I'm doing with iOS

whitedogg13, 2020-01-22:
@jberns88 No worry! I'd like to make the library better, so I'm actually appreciated for those in-depth issues you reported, though I might not be able to feedback or fix them quickly.

Regarding the `Ndef + NfcV` case, isn't the `NTAG213` family belong to `MifareUltralight` product line? I do have several `NTAG213` tags, and those can be detected by using `Mifare` tech in iOS. Apple does say `ultralight` and `desfire` are compatible with `NFCMifareTag`, though `classic` is not mentioned.

About the second issue `Cancel not rejecting promise`, I will investigate it and update to you later.

Thanks!

jberns88, 2020-01-22:
> isn't the NTAG213 family belong to MifareUltralight product line?

That might be true. TBH I was ignorantly assuming it was only Ndef as I wasn't aware this was a thing. Will test this tomorrow. Thanks.

jberns88, 2020-01-23:
So you was 100% right and I was 100% wrong!

Detecting Mifare then reading the Ndef works perfectly. 🎉 

Thanks you.

Do you think it's worth throwing an error if someone is trying to detect Ndef + other instead of silently ignoring the other tech? It might stop people like me falling into the same trap or maybe remove Ndef from ```requestTechnology()``` and have a separate ```requestNdef()``` API.

Happy to do a PR for this as it's all JS based changes if you think one of the solutions is right.

whitedogg13, 2020-01-24:
@jberns88 I think we can stick to the API, but add a warning message when NDEF and specific tech type is mixed together in iOS. That will be a simpler alternative without changing the API behavior but also give hints to developers, since it’s still a valid use for Android. If you can make a PR for that it will be great!

Besides that, I’m considering to enhance the `getTag` call in iOS, to perform a tech type enumeration just like Android, (which does this natively)  to list the supported techs for the tag once it’s scanned. In this way, we no longer need to register for specific tech types.

JimTeva, 2020-01-27:
Thank you @whitedogg13  for your quick feedback, it is greatly appreciated!

JimTeva, 2020-01-27:
@jberns88 should you close this issue?

jberns88, 2020-01-27:
Just waiting for @whitedogg13 to look the **Cancel not rejecting promise** issue.

whitedogg13, 2020-01-28:
Hi @jberns88 I just fixed this, and published `v2.1.0` as an official release. Please check it out!

jberns88, 2020-01-28:
Thanks @whitedogg13 for all the changes/fixes. I've got everything I need to complete my project now :)

closed 2020-01-28 by jberns88