revtel/react-native-nfc-manager #237: can only read ID from tag

ThePF, 2020-01-16:
Hey, I copied the v2-ios+android-read-ndef version and when I scan the tag I get tag {"id": "AEA5B0", "techTypes": ["android.nfc.tech.NfcV", "android.nfc.tech.NdefFormatable"]} 
When I copied the v2-ios+android-get-uid version I get "ex fail to connect tag"
How can I read the actual memory of the tag? 
Thanks for the help

jberns88, 2020-01-17:
You need to use transceive for nfcv. The Ndef example is the wrong one. 

You need to know what baudrate, command you want to send, and the memory block you want to read.

```
NfcManager.transceive([0x02, 0x20, 0x00]);
```
0x02 = high baudrate
0x20= read command
0x00 memory block 0

whitedogg13, 2020-01-21:
@ThePF From the `techTypes` reported above, we can see the `android.nfc.tech.Ndef` is not in list, which means you tag contains no valid NDEF messages. However, since `android.nfc.tech.NdefFormatable` is listed, so the tag is actually NDEF compatible just has no data.

ThePF, 2020-02-10:
Hey, thanks for the help so far. I still didn't figured out how to call the transceive method. I know now: 

baudrate: 9600
Blocksizes: 8 Bytes
memory block 2 - 8 

Also can't find any example in your documentations on how reading nfcV with transceive, or did I miss it?

lukebars, 2021-02-01:
@whitedogg13 is there any example how to use NfcV transceive?

closed 2021-05-26 by github-actions[bot]