revtel/react-native-nfc-manager #242: The Button to scan works but It can not read  (ios)

BlastTheSponge, 2020-01-24:
Hello, 

I've got a problem with IOS only. When my button is clicked to start the NFC reading, this image appears.
[image]

But it cannot read anything.
When I try with another application it works.

The NFC tag reader in capabilities is here, the entitlement is correct (with NDEF and TAG). My signing is correctly linked too.

The Android version is working too.

Thanks

Frans-L, 2020-01-26:
Without any code hard to help. Could you make a simple example where that possible bug happens?

BlastTheSponge, 2020-01-27:
There is my "NfcIos" code with the button

```
import React from 'react';
import {View, StyleSheet, TouchableOpacity} from 'react-native';
import NfcManager from 'react-native-nfc-manager';

import Typography from './Typography';
import theme from '../utils/theme';
import readTag from '../utils/read-tags';

export default class NfcIOs extends React.PureComponent {
  state = {
    show: true,
  };

  componentDidMount = async () => {
    try {
      await NfcManager.start();
      this.setState({show: true});
    } catch (error) {
    }
  };

  componentWillUnmount = () => {
    NfcManager.unregisterTagEvent();
    NfcManager.stop();
  };

  handleScanPress = () => {
    NfcManager.registerTagEvent(
        this.handleTagDetected,
        'Balayez la partie supérieure du lecteur',
        true,
    );
  };

  handleTagDetected = (tag) => {
    const {onNfcRead, navigation, onNfcInitialized} = this.props;
    try {
      if (onNfcRead) {
        onNfcRead();
      }
      readTag({tag, navigation});

      onNfcInitialized && onNfcInitialized();
    } catch (err) {
      onNfcInitialized && onNfcInitialized();
    }
  };

  render() {
    const {show} = this.state;

    return show ? (
        <View style={style.container}>
          <TouchableOpacity style={style.button} onPress={this.handleScanPress}>
            <Typography style={style.text}>Badgez avec votre iPhone</Typography>
          </TouchableOpacity>
        </View>
    ) : null;
  }
}
```

closed 2021-05-26 by github-actions[bot]