revtel/react-native-nfc-manager #243: Android - registerTagEvent not triggering callback

jhoward1994, 2020-01-27:
Hi,

Please can you help with the below issue?

I have implemented **NfcManager.registerTagEvent** inside a **_startDetection()** method in a couple of different ways:

    NfcManager.registerTagEvent(
      tag => {
        console.log("Tag Discovered", tag);
      },
      "Hold your device over the tag",
      {
        invalidateAfterFirstRead: true,
        isReaderModeEnabled: true,
        readerModeFlags:
          NfcAdapter.FLAG_READER_NFC_A | NfcAdapter.FLAG_READER_SKIP_NDEF_CHECK
      }
    );

    NfcManager.registerTagEvent(this._onTagDiscovered)
      .then(result => {
        console.log("registerTagEvent OK", result);
      })
      .catch(error => {
        console.warn("registerTagEvent fail", error);
      });

In the first case, the console log is never triggered even when I scan an NFC tag 
In the second case, the _onTagDiscovered callback is never triggered and I see the log of "registerTagEvent OK" immediately after starting detection.

This issue seems similar to - https://github.com/whitedogg13/react-native-nfc-manager/issues/136

Any help would be greatly appreciated

Thanks,
Jamie

whitedogg13, 2020-01-31:
Hi @jhoward1994 Sorry for the late response.

Looks like you're using the deprecated API signature. We have several breaking change from v1 => v2, please try to use [this example](https://github.com/whitedogg13/react-native-nfc-manager/blob/master/example/AppV2.js)

overlord117, 2020-06-01:
Hello,

I am having the exact same issue and not sure how to resolve it. I haven't tested this will iOS yet, but on Android the onTagDiscovered callback is never triggered. In regards to the v2 example, the registerTagEvent doesn't seem to wait for a tag to be read. Logging in the console shows that the _test method runs without ever reading a tag once the Test button is pressed. Any guidance would be much appreciated.

Thanks,
Alex

closed 2021-05-26 by github-actions[bot]