revtel/react-native-nfc-manager #246: Expose iOS NFCReaderSession invalidate() method

JimTeva, 2020-01-29:
react-native-nfc-manager v2.1.0

@whitedogg13 I'm trying to close the iOS default alert as soon as the success icon finishes.

From that [source](https://itnext.io/swift-corenfc-as-quick-as-possible-79544796cd0a), it looks like I need to do 2 things:

1 - set invalidateAfterFirstRead to false on requestTechnology:
```js
NfcManager.requestTechnology(nfcTechs, {
      alertMessage: 'Scan a tag!',
      invalidateAfterFirstRead: false,
    })
```

2 - use invalidate() [method](https://developer.apple.com/documentation/corenfc/nfcreadersessionprotocol/2874107-invalidate)

Right now, there is delay of around 2.5s between the success icon finishes to draw and the alert is dismissed.

I've checked NFCManager.js and it looks like the method is not exposed, can you add it please?

Thanks!

whitedogg13, 2020-01-29:
@JimTeva sure, I will add this method soon.

whitedogg13, 2020-01-31:
Hi @JimTeva the `invalidateSessionIOS` has been added in `v2.1.1`, thanks!

JimTeva, 2020-02-01:
Hi @whitedogg13 , thank you so much for adding it so fast!
Unfortunately, I just found out that the method is useless in my case.
Indeed I am trying to start a session with techs other than Ndef. The result is that it is registering with NFCTagReaderSession in native. But the property "invalidateAfterFirstRead" is only working when initializing a NFCNDEFReaderSession......
To sum up: To close the iOS default alert as soon as the success icon finishes, we can (at the moment) only use NDef Tech.
I am closing this issue. Thanks Richie

closed 2020-02-01 by JimTeva

whitedogg13, 2020-02-02:
@JimTeva Got it, thanks for the insight!

Paul-Lategan, 2020-04-15:
Hi @JimTeva & @whitedogg13, sorry if I'm missing something obvious in the thread above, but is there a function to call to immediately close the NFC dialog (with the checkbox) that appears for 3 seconds after scanning an NFC tag?

I'm trying to do this: https://stackoverflow.com/a/50730560/972191.

My relevant code below:

`export const cleanUpNFC=function() {
    NfcManager.cancelTechnologyRequest().catch(() => 0);
    NfcManager.setEventListener(NfcEvents.DiscoverTag,null);
    }

export const closeNFCDialog=function() {
    // TODO: close NFC dialog immediately here
    }

export const getNFCTag=async (hlptxt, cbk) => {
    // register what the NFC read event will do:
    await NfcManager.setEventListener(NfcEvents.DiscoverTag,tag => {
        NfcManager.setAlertMessageIOS("Tag has been read. One moment...");

        // more on parsing here: https://reactnativeexample.com/a-nfc-module-for-react-native/
        let parsed=null;

        if(tag.ndefMessage && tag.ndefMessage.length > 0) {
            // ndefMessage is actually an array of NdefRecords,
            // and we can iterate through each NdefRecord, decode its payload
            // according to its TNF & type
            const ndefRecords=tag.ndefMessage;

            function decodeNdefRecord(record) {
                if(Ndef.isType(record, Ndef.TNF_WELL_KNOWN, Ndef.RTD_TEXT)) {
                    return ["text",Ndef.text.decodePayload(record.payload)];
                    }
                else if(Ndef.isType(record, Ndef.TNF_WELL_KNOWN, Ndef.RTD_URI)) {
                    return ["uri",Ndef.uri.decodePayload(record.payload)];
                    }

                return ["unknown","---"]
                }

            parsed=ndefRecords.map(decodeNdefRecord);
            }

        try {
            cbk(parsed[0][1]);
            }
        catch(err) {
            console.error(err);
            alert("(empty tag)");
            }

        NfcManager.unregisterTagEvent().catch(() => 0);
        });

    // trigger an NFC read attempt:
    try {
        await NfcManager.registerTagEvent();
        NfcManager.setAlertMessageIOS(hlptxt);
        }
    catch(err) {
        console.warn("err",err);
        NfcManager.unregisterTagEvent().catch(() => 0);
        }
    }

export const initNFC=function() {
    NfcManager.start();
    }

export const setNFCTag=async (wrttxt, hlptxt) => {
    try {
        let resp=await NfcManager.requestTechnology(NfcTech.Ndef, {
            alertMessage : (hlptxt || "")
            });

        let bytes=Ndef.encodeMessage([Ndef.textRecord(NFC_API_V1_PREFIX + wrttxt)]);
        await NfcManager.writeNdefMessage(bytes);

        NfcManager.setAlertMessageIOS("Writing NFC Tag. One moment...");
        NfcManager.cancelTechnologyRequest().catch(() => 0);
        }
    catch(err) {
        console.error("err",err);
        NfcManager.unregisterTagEvent().catch(() => 0);
        }
    }`

JimTeva, 2020-04-15:
Hi @Paul-Lategan 

I tried that method too (the stackoverflow link you provided).
Looks like you are working on NDEF tech so you might have better luck than me hehe. (fyi, in iOS, the property "invalidateAfterFirstRead" is only working when initializing a NFCNDEFReaderSession)

1 - No need to set invalidateAfterFirstRead to false on requestTechnology as it is the default value in nfcmanager.js

2 - @whitedogg13 has added method invalidateSessionIOS so you should be able to use it with NfcManager object now: I haven't test it but you should try NfcManager.invalidateSessionIOS()

You should have much more info by going through @whitedogg13 code [here](https://github.com/whitedogg13/react-native-nfc-manager/blob/master/NfcManager.js).

Paul-Lategan, 2020-04-17:
@JimTeva thanks mate!  I upgraded to 2.1.6 and called NfcManager.invalidateSessionIOS() which works perfectly.  @whitedogg13 thank you for this awesome lib.