revtel/react-native-nfc-manager #247: NFCManager Null is not an object

jeneiro, 2020-01-29:
I am trying to start a project with the react-native-nfc-manager. 
I keep getting this error:

TypeError: TypeError: null is not an object (evaluating 'NativeNfcManager[name]')
- node_modules\react-native-nfc-manager\NativeNfcManager.js:11:40 in callNative
* components\nfcscanner\nfcscanner.js:49:4 in componentDidMount

Here is my code below:
import React, { Component } from "react";
import {
  TouchableOpacity,
  Platform,
  ImageBackground,
  BackHandler,
  ScrollView,
  I18nManager,
  AsyncStorage,
  SafeAreaView,
  TextInput,
  Alert,
  StyleSheet
} from "react-native";
import NfcManager, { NfcTech } from "react-native-nfc-manager";
import {
  Content,
  Container,
  Form,
  Input,
  Header,
  Left,
  Body,
  Title,
  Segment,
  Label
} from "native-base";

import styles from "./styles";

export default class NFCScanner extends Component {
  constructor(props) {
    super(props);
    this.state = {
      log: "Ready...",
      text: ""
    };
  }
  componentDidMount() {
    NfcManager.start();
  }

whitedogg13, 2020-01-31:
@jeneiro looks like it lacks native part, so the NativeNfcManager is evaluated to null. 

Are you using Expo? If so, you will need to eject from expo first.

jeneiro, 2020-01-31:
I am using expo. does that mean I can't use Expo as a CLI while developing my NFC component?

whitedogg13, 2020-01-31:
If you have to use Expo as CLI for some reason, you will have to eject from it, see [here](https://docs.expo.io/versions/latest/expokit/eject/).

Expo only allows pure JS dependencies, not native ones. When installing a library with native part, it won't be able to retrieve corresponding native module in runtime.

I will recommand you to use `react-native-cli`  when a 3rd party library with native dependency is required. Some info can be found [here](https://facebook.github.io/react-native/docs/getting-started), please choose the `React Native CLI Quickstart` tab.

closed 2020-02-02 by jeneiro