revtel/react-native-nfc-manager #248: Isodep on ios errors with "No tech matches"

khpa, 2020-01-29:
Hi, im trying to run the AppV2Apdu example on ios (13.3 iphone 8) but get "No tech matches" when a device is placed on the phone. The device is an NXP (ISO14443-4), it works on android fine.

2020-01-29 13:55:38.219348+0000 ManageMii[10743:3198487] NFCTag didBecomeActive
2020-01-29 13:55:43.018625+0000 ManageMii[10743:3198487] NFCTag didDetectTags
2020-01-29 13:55:43.031 [info][tid:com.facebook.react.JavaScript] 'errors = ', 'No tech matches'

whitedogg13, 2020-01-31:
hi @khpa , this means iOS doesn't consider the tag to have IsoDep tech. For iOS, it will automatically select AID for you, and you must write the AID down in your info.plist. However in Android, it simply gives you all capabilities to communicate with the NFC chip through `transceive` directly.

The ios part for AppV2Apdu.js expects a e-passport supported NFC chip, and you need to add following lines into you info.plist:
```
<key>com.apple.developer.nfc.readersession.iso7816.select-identifiers</key>
<array>
    <string>A0000002471001</string>
</array>
```

khpa, 2020-02-27:
hi @whitedogg13 sorry for the late reply, this worked, the AID is mandatory in info.plist, thanks again your help!

closed 2020-02-27 by khpa