revtel/react-native-nfc-manager #249: NfcManager.getTag() fails with NfcTech.Ndef

acalderaro, 2020-02-04:
hello,  i am trying to read the id and message using this code:
`
read = async () => {
      try {
        let tech = Platform.OS === 'ios' ? NfcTech.Ndef : NfcTech.NfcA;
        let resp = await NfcManager.requestTechnology(tech, {
          alertMessage: 'Ready to do some custom Mifare cmd!'
        });
        console.log({ resp });

        // the NFC uid can be found in tag.id

        let tag = await NfcManager.getTag();
        console.log({ tag });

        // then read NDEF message
        let ndef = await NfcManager.getNdefMessage();
        console.log({ ndef });

        await NfcManager.cancelTechnologyRequest();
      } catch (ex) {
        console.log(ex);
      }
    };
`

but i get "Not even registered" error. any idea of why?

whitedogg13, 2020-02-04:
@acalderaro In iOS, we won't be able to get `id` when requesting `NfcTech.Ndef` technology, that's a platform limitation. 

To get the `id` in iOS, our best bet would be something like:
```javascript
// for iOS
await NfcManager.requestTechnology(
  [NfcTech.MifareIOS, NfcTech.Iso15693IOS, NfcTech.IsoDep], 
  {  alertMessage: 'Ready to do some custom Mifare cmd!'  }
);
let tag = await NfcManager.getTag();
console.log({ tag }); // tag object should contain `id` now!
```

acalderaro, 2020-02-04:
@whitedogg13 thank you for your answer.

closed 2020-02-04 by acalderaro