revtel/react-native-nfc-manager #250: Examples not working

desteve, 2020-02-04:
Hi everybody.

Just wanted to comment that most of the examples are not working, some of them can't even been built. Some work on Android but not in iOS. The version of the library to use with them is not clear. All these issues make the library hard to use and a time-consuming one, wasting hours for some easy test.
I think you should reconsider offering ready-to-go examples.

Regards.

whitedogg13, 2020-02-04:
Hi @desteve , not sure what issues you encountered or what examples break through your comment. 

In NFC domain, most of the real use case involves specific NFC technology communication (Mifare, IsoDep or NfcV), and they’re generally domain specific and pretty complex, so we are not sure what kind of “ready-to-go” example to offer. Maybe you can tell us what example you expect, so we can make one for that.

desteve, 2020-02-04:
Hi.

Sure, there are a lot of issues involving NFC technology, but I expect some simple clear information about library usage. In fact, I just expect something as simple as AppV2.js, but working on both platforms (iOS and Android).

Thanks.

whitedogg13, 2020-02-04:
For me, the AppV2.js works for both platforms. 

Please notice that, in Android, there's no built-in NFC scan UI, so there won't be any visual cue after you tap the "Test" button, but once a tag is scanned, there will be a `console.warn` message.

Hope that helps.

desteve, 2020-02-04:
Hi.
On Android it's working fine at first shot. Not in iOS, other apps are reading my tags perfectly but this is not happening with the library and this example. I'm driving nuts myself to get it working, changing library versions, detection flow, but I'm unable to put it to work.

Thanks.

whitedogg13, 2020-02-05:
Hi @desteve for iOS, there're some extra configurations required. 
First, we need to enable the NFC capability in Apple developer site:
[image]

Also need to add the app's entitlement through Xcode:
[image]

Then we also need to add NFC privacy statement into Info.plist:
[image]

I have put a [repo](https://github.com/whitedogg13/nfc-test-app) for your reference. Please feedback if there're any other issues.

desteve, 2020-02-05:
Hi, thanks for your help.

I had all that configuration already done. In fact, iOS app starts to scan for NFC Tags (a system dialog appears reading to tap a tag), but no tag is detected, no matter which type. Same tags, are detected at first shot with different NFC reading apps available on AppStore.

Regards.

whitedogg13, 2020-02-06:
Sorry to hear that, but that's all setup required at least in my case. I have tested on iPhone 8, XR, and 11.

Not sure what I can do to help, maybe you can try [this repo](https://github.com/whitedogg13/nfc-test-app) to see if it works...

desteve, 2020-02-06:
Thanks for your help. I'll try the example you provide.
I'm closing the issue because I'm in some kind of dead-end which I have to solve by myself.

Thanks!

closed 2020-02-06 by desteve

desteve, 2020-02-06:
I get the example working finally! Still don't know what was the problem, but it's working.

Thanks for your help.