revtel/react-native-nfc-manager #251: iOS 13.3 not working.

jbejas, 2020-02-04:
Hello guys!

I've set up a React Native app to read NFC tags but it doesn't seem to work. The Reader dialog shows up, but nothing gets read.

Here is my code.

```
import React, { Component } from "react";
import { View, TouchableOpacity, Dimensions, Image, Platform } from 'react-native';
import NfcManager, {NfcEvents} from 'react-native-nfc-manager';
class RFIDREaderScreen extends Component {
  componentDidMount() {
    NfcManager.start();
    NfcManager.setEventListener(NfcEvents.DiscoverTag, tag => {
      console.warn('tag', tag);
      this.setState({ text: tag });
      NfcManager.setAlertMessageIOS('I got your tag!');
      NfcManager.unregisterTagEvent().catch(() => 0);
    });
  }

  componentWillUnmount() {
    NfcManager.setEventListener(NfcEvents.DiscoverTag, null);
    NfcManager.unregisterTagEvent().catch(() => 0);
  }
  _cancel = () => {
    NfcManager.unregisterTagEvent().catch(() => 0);
  }

  _test = async () => {
    try {
      await NfcManager.registerTagEvent();
    } catch (ex) {
      console.warn('ex', ex);
      NfcManager.unregisterTagEvent().catch(() => 0);
    }
  }
}
```

I'm not sure what I'm missing here...

whitedogg13, 2020-02-05:
Hi @jbejas ,

For iOS, there're some extra configurations required. 
First, we need to enable the NFC capability in Apple developer site:
[image]

Also need to add the app's entitlement through Xcode:
[image]

Then we also need to add NFC privacy statement into Info.plist:
[image]

I have put a [repo](https://github.com/whitedogg13/nfc-test-app) for your reference. Please feedback if there're any other issues.

yassine-be, 2020-02-11:
Hi @whitedogg13 , same as @jbejas , the v2 example not working on iphone7 and iphoneXS IOS 13.3.1
I have as you mentioned activated the NFC capability and added the permissions needed.
the app show the reading screen, but it's not detecting anything.

whitedogg13, 2020-02-11:
Hi @yassine-be please try to use [this example](https://github.com/whitedogg13/nfc-test-app) to see if it works.
(in issue #250, some developer reports the issue been fixed by trying above example, although the root cause is not identified)

yassine-be, 2020-02-12:
the example does not seem to work for me ( though working on android ).
can someone @whitedogg13 and @jbejas please confirm if scan is working on ios 13.3.1 for them.
i tried another app ("decode" from apple store) and it is not scanning either.

whitedogg13, 2020-02-13:
Hi @yassine-be The example does work for me, on iOS 13.3 + iPhone 11. 

Can you check if your entitlement is correct like this?

[image]

(This file is automatically generated when we add our NFC capability in Xcode's `Signing & Capabilities` tab)

yassine-be, 2020-02-17:
thanks @whitedogg13 for your help, the example is working for me now.
it turns out the tags i was testing with were not working on ios. I got some new ones and it's going smooth now.
i'm not sure as to why they worked only for android. I'll update my post if i find out.

closed 2021-05-26 by github-actions[bot]