revtel/react-native-nfc-manager #253: Don't know how to use the transceive method

ThePF, 2020-02-19:
hey everyone, I still didn't figured out how to call the transceive method. I know now:

baudrate: 9600
Blocksizes: 8 Bytes
memory block 2 - 8

Also can't find any example in the documentations on how reading nfcV with transceive, or did I miss it?

whitedogg13, 2020-02-26:
Hi @ThePF , sorry for the late response!

it depends on your platform. If you're in `iOS`, you can use `readSingleBlock` to read the memory block from NFCV tags. For `Android`, you will need to compose the low-level transceive bytes youself.

ThePF, 2020-02-26:
Hey @whitedogg13 thank you for your response. I still don’t know how to do any of these two, what would be the correct syntax? also can’t find any example that works for me.. could you link me to one please.

whitedogg13, 2020-02-27:
Hi @ThePF For iOS, you can first look at [this example](https://github.com/whitedogg13/react-native-nfc-manager/blob/master/example/AppV2Iso15693.js). I will try to give you an Android example later when I'm available.

ThePF, 2020-03-06:
Hey @whitedogg13  I'm actually looking for a solution for Android, since as far as I know there are some restraints for IOS. It would be great if you could send me an example for Android. Thank you very much!

Amanda1995pig, 2021-01-28:
I'm  looking for a solution for Android too .Are you resolve it ?

lukebars, 2021-02-02:
@whitedogg13 is there an example how to read NfcV tag? iOS works fine, but android doesn't work at all and there is no documentation on that.

closed 2021-05-26 by github-actions[bot]