revtel/react-native-nfc-manager #254: TypeError undefined is not an object

Quintinator, 2020-02-20:
So while developing my app I constantly run into the same error.
`TypeError: TypeError: Undefined is not an object (evaluating'_reactNativeNfcManager.NfcEvents.DiscoverTag')`

To start I have ejected my project using "Expo eject"
I have linked it
I start the app using "Npx react-native run-android" while my android phone is connected through USB.

The code is from the read exampe V2. 
`import React, { Component } from 'react';
import {     
    SafeAreaView,
    StyleSheet,
    View,
    Text,
    Alert,
    Platform,
    Image
 } from 'react-native';
import Dimensions from "Dimensions";
import { TextInput, TouchableOpacity } from 'react-native-gesture-handler';
import axios from "axios";
import NfcManager, {NfcEvents} from 'react-native-nfc-manager'

export default class NFCScreen extends React.Component {
    

    static navigationOptions = {
    };
    constructor(props) {
        super(props);
        this.state = {
            log: "Ready...",
            text: ""
        };
    }
    
    componentDidMount() {
        NfcManager.start();
        NfcManager.setEventListener(NfcEvents.DiscoverTag, tag => {
          console.warn('tag', tag);
          NfcManager.setAlertMessageIOS('I got your tag!');
          NfcManager.unregisterTagEvent().catch(() => 0);
        });
      }
    
      componentWillUnmount() {
        NfcManager.setEventListener(NfcEvents.DiscoverTag, null);
        NfcManager.unregisterTagEvent().catch(() => 0);
      }

    render(){
        const { navigate } = this.props.navigation;
        return (
            <View style={styles.container}>
                
                <View style={styles.knopView}>
                    <TextInput style={styles.handmatig}
                        returnKeyLabel='Done'
                        returnKeyType='done'
                        placeholder="Handmatig invoeren..."
                        onChangeText={(id) => this.setState({ id })}
                        value={this.state.id}

                    /*Code voor handmatige invoer, voor mensen zonder NFC-schip in hun telefoon*/
                    />

                    <TouchableOpacity style={styles.knop} onPress={this._handlePress.bind(this)}>
                        <Text style={styles.handmatigKnop}>OK</Text>
                    </TouchableOpacity>
                </View>

                <Image style={styles.nfcImage}
                    resizeMode="contain"
                    source={require("../../assets/images/nfcfc.png")}
                /*NFC logo*/
                />
                <Text style={styles.nfcText}>NFC</Text>
                <TouchableOpacity style={styles.scanKnop} onPress={this._test}/*Code voor de scan knop, bij ios moet deze ingedrukt worden*/>
                    <Text style={styles.scanText}>Scan NFC.</Text>
                </TouchableOpacity>
                <TouchableOpacity style={styles.scanKnop} onPress={this._cancel}>
                    <Text style={styles.scanText}>Cancel</Text>
                </TouchableOpacity>

            </View>
        );
        }

    async _handlePress(event) {
        const { navigate } = this.props.navigation;
        let manualId = this.state.id;
        try {
            
            // the API call
            let result = await axios.get(
                `https://api.concremote.com/Version/v1/factoryproductioncontrol/product?$filter=RFID%20eq%20'${manualId}'&$top=1&$expand=ProductType`);
            
            //shows what is in the array
            console.log(result.data.value)

            //Should technically show the Id that has been shown in the array.
            //console.log(result.data.value)
            const product = result.data.value[0];
            console.log('Product: ', product);
            console.log('Product id: ', product.Id);

            navigate("ProductScreen", {
                id: product.ProductType.Id,
                name: product.ProductType.Name + " " + product.Code
              });

        //Catch of the error
        } catch (err) {
            console.log(err);
            Alert.alert('Waarschuwing!', 'Het ingevoerde ID is onjuist.', [{text: 'Ok'}]);
         }
        // This shows which ID has been filled in.
        console.log(manualId);
    }

    _cancel = () => {
        NfcManager.unregisterTagEvent().catch(() => 0);
      }

    _test = async () => {
        try {
          await NfcManager.registerTagEvent();
        } catch (ex) {
          console.warn('ex', ex);
          NfcManager.unregisterTagEvent().catch(() => 0);
        }
      }

}
const DEVICE_HEIGHT = Dimensions.get("window").height;
const DEVICE_WIDTH = Dimensions.get("window").width;

const styles = StyleSheet.create({
    container: {
        flex: 1,
        alignItems: "center",
        //justifyContent: "center"
    },
    nfcImage: {
        height: DEVICE_HEIGHT / 3.5,
        marginBottom: 17,
        opacity: 0.6,
        alignSelf: "center",
        justifyContent: "center",
        marginTop: DEVICE_HEIGHT / 8
        /*Styling voor het NFC logo*/
    },
    nfcText: {
        fontSize: 60,
        color: "#1b395b",
        fontWeight: "bold",
        opacity: 0.6,
        alignSelf: "center"
        /*Styling voor de tekst onder het logo*/
    },
    handmatig: {
        borderBottomWidth: 1.5,
        width: DEVICE_WIDTH / 2.2,
        fontSize: 15,
        alignSelf: "center",
        marginTop: 15,
        marginBottom: 15
        /*Styling voor de handmatige invoer*/
    },
    handmatigKnop: {
        fontSize: 20,
        color: "white"
    },
    knop: {
        marginLeft: 15,
        textAlign: "center",
        alignSelf: "center",
        backgroundColor: "#297ef5",
        borderWidth: 8,
        borderColor: "#297ef5",
        borderRadius: 10
    },
    knopView: {
        marginTop: 11,
        width: DEVICE_WIDTH / 1.65,
        alignSelf: "center",
        flexDirection: "row"
    },
    scanKnop: {
        alignSelf: "center",
        backgroundColor: "#297ef5",
        borderWidth: 10,
        borderColor: "#297ef5",
        borderRadius: 12,
        marginTop: DEVICE_HEIGHT / 20
        /*Styling voor de scanknop*/
    },
    scanText: {
        color: "white",
        fontSize: 20
        /*Styling voor de tekst in de scanknop*/
    }
});
`

Quintinator, 2020-02-25:
Tried some more but still haven't figured it out.

whitedogg13, 2020-02-26:
Hi @Quintinator sorry for the late response.

I'm not using expo, but since this module contains native part, it will be easier to use `react-native-cli` to create project skeleton. 

For example:
```bash
npx react-native init NfcTestApp
cd NfcTestApp
npm i react-native-nfc-manager
```

Or, you can use this [repo](https://github.com/whitedogg13/nfc-test-app) as a starting point.

Quintinator, 2020-02-27:
I'll try that. If that works I know it's due to issues with my project. Some other modules aren't working either

Quintinator, 2020-02-28:
It works now. Project was configured through expo. Causing it not to work.

closed 2021-05-26 by github-actions[bot]