revtel/react-native-nfc-manager #255: writing and reading at the same time, is it possible?

BravenxX, 2020-02-21:
Is it possible to write and read at once with this library?? I didn't see an example in the documents about this.

whitedogg13, 2020-02-21:
Yes, it's possible like this:
```javascript
import NfcManager, {NfcTech, Ndef} from 'react-native-nfc-manager';
...
writeTextNdef = async (text) => {
    try {
      await NfcManager.requestTechnology(NfcTech.Ndef);

      // read the tag, and NDEF will be in `tag.ndefMessage`
      let tag = await NfcManager.getNdefMessage();

      // write the tag
      let bytes = Ndef.encodeMessage([
        Ndef.textRecord(text),
      ]);
      await NfcManager.writeNdefMessage(bytes);
      }
    } catch (ex) {
      console.warn(ex);
    }

    NfcManager.cancelTechnologyRequest();
    return originalTag;
}
```

BravenxX, 2020-02-25:
> Yes, it's possible like this:
> 
> ```js
> import NfcManager, {NfcTech, Ndef} from 'react-native-nfc-manager';
> ...
> writeTextNdef = async (text) => {
>     try {
>       await NfcManager.requestTechnology(NfcTech.Ndef);
> 
>       // read the tag, and NDEF will be in `tag.ndefMessage`
>       let tag = await NfcManager.getNdefMessage();
> 
>       // write the tag
>       let bytes = Ndef.encodeMessage([
>         Ndef.textRecord(text),
>       ]);
>       await NfcManager.writeNdefMessage(bytes);
>       }
>     } catch (ex) {
>       console.warn(ex);
>     }
> 
>     NfcManager.cancelTechnologyRequest();
>     return originalTag;
> }
> ```

Thanks!

closed 2020-02-26 by whitedogg13

Damian333, 2021-02-23:
Is it possible to on one swipe read Card > check if one of techTypes is android.nfc.tech.NdefFormatable and if is format card> and then write? 
Is it possible both on Android and IOS?

whitedogg13, 2021-02-23:
@Damian333 Yes, It should be possible, but we will have to implement this for different NfcTechs.

For the most used NfcA (Mifare) tags, we have already implemented in [this project](https://github.com/revtel/react-native-nfc-rewriter), you can search for `NfcProxy.js` which has an `eraseNfcA` function does this.

swiersyoram, 2022-11-03:
Hi @whitedogg13 is it possible to change technology type between reading and writing? I for example want to read mifareUltralightReadPages and write an ndef message? Thanks for the great lib btw!