revtel/react-native-nfc-manager #256: Example reading/writing from separate devices

r0fls, 2020-02-21:
Thank you so much for this library @whitedogg13, it's very useful! However I'm having trouble getting any of the examples to work 😄 Is there any plan to add an example with two separate devices?

This is what I'm doing:

**reader**

```
  const handleTestNdef = async () => {
    try {
      console.log("waiting")
      await NfcManager.requestTechnology(NfcTech.Ndef);
      // I've tried this approach too
      //resp = await NfcManager.registerTagEvent()
      //resp = await NfcManager.setEventListener(NfcEvents.DiscoverTag, tag => {
      //  console.warn('tag', tag);
      //  }
      let ndef = await NfcManager.getNdefMessage()
      // Never gets here
      console.log("done waiting again")
      console.log(ndef)
    }   
    catch (ex) {
      console.warn('ex', ex) 
    }   
  }
```

**writer**

```
  const handleNext = async () => {
    try {
      var resp = await NfcManager.registerTagEvent()
      let bytes = Ndef.encodeMessage([
        Ndef.textRecord("i am a string"),
      ]); 
      let r = await NfcManager.writeNdefMessage(bytes)
      console.log(r)
    } catch (err) {
      console.log("error", err)
    }   
  }
```

The one trying to pay logs: `  ex No ndef available` and the one trying to send the message logs: `error Not connected`. 

I am also calling ` NfcManager.start()` in my component did mount lifecycle hook and `NfcManager.cancelTechnologyRequest()` on unmount.

Do you see what I'm doing wrong? Any help would be appreciated.

r0fls, 2020-02-21:
Also for the record I'm using v2.0.3 of the library

whitedogg13, 2020-02-22:
Hi @r0fls not sure the actual task you want to perform. Do you want to exchange data between two devices?

r0fls, 2020-02-22:
Yes, exactly!

egislook, 2020-02-25:
Yes, I'm trying to send a message from one device to another and its a bit confusing. Would be nice to see what's the issue with above code

egislook, 2020-02-26:
@r0fls @whitedogg13 can you even make one phone as sender and another one as a receiver? We are trying to apply this method to send short information but documentation is very confusing.

whitedogg13, 2020-02-26:
Hi all, sorry for the late response! A bit busy recently.

Transfering data directly among devices is only available for `Android` using NFC `P2P` mode, and here is a basic example:
```javascript
// sender
import NfcManager, {Ndef} from 'react-native-nfc-manager';
...
function sayHelloToPeer() {
  let bytes = Ndef.encodeMessage([
    Ndef.textRecord('hello'),
  ]);
  NfcManager.setNdefPushMessage(bytes);
}

// on receiver side, you can use `example/AppV2` to check the received NDEF
```
Notice in Android, it's also called `Android Beam`.

Some extra points:
1. `iOS` doesn't expose NFC P2P mode API, so we cannot exchange data this way.
2. Currently we can only send data in one direction, which means the received peer cannot generate any response payload.
3. Some report says this feature will be deprecated in Android Q, see [this link](https://www.xda-developers.com/google-deprecate-android-beam-api-nfc-share-files/) for more detail.

r0fls, 2020-02-26:
Is there any reason this isn't implemented for iOS? I see it should be supported here: 

>Your app can also write data to tags, and interact with protocol specific tag such as ISO 7816, ISO 15693, FeliCa™, and MIFARE® tags.

https://developer.apple.com/documentation/corenfc#topics
https://developer.apple.com/documentation/corenfc/creating_nfc_tags_from_your_iphone

egislook, 2020-02-27:
Sad that we can not just create an NFC tag out of our phone. That would solve the problems. For me, Android solution is enough at this stage but it seems like this technology is forcing you to use tags :D that does not make any sense when you can use phones :D

egislook, 2020-02-27:
Ok @r0fls and @whitedogg13 thank you. We have a working example using Android Beam
#Sender
```
import NfcManager, {Ndef, NfcEvents} from 'react-native-nfc-manager';
React.useEffect(() => {
    NfcManager.start()
    const bytes = Ndef.encodeMessage([ Ndef.textRecord('===Message to receiver') ])
    // Stores the message in Senders phone. Now any other android with enabled NFC can receive it
    NfcManager.setNdefPushMessage(bytes)
}, [])
```
#Receiver
```
import NfcManager, {Ndef, NfcEvents} from 'react-native-nfc-manager';
React.useEffect(() => {
    NfcManager.start()
    NfcManager.registerTagEvent()
    // Listens for NFC when u put phones together and tap the screen
    NfcManager.setEventListener(NfcEvents.DiscoverTag, ({ ndefMessage: [{ payload }]}) => {
      const [{ type }] = Ndef.decodeMessage(payload)
      console.log(type) //Message u should receive from sender
    })
}, [])
```

I have one question about how we can confirm the beam NFC without taping. Is there any way to make it instant for Sender?

whitedogg13, 2020-02-27:
@r0fls The iOS only supports NFC reader/writer mode, not p2p mode. 

In short, the reader mode requires two different roles to work, initiator (phone) VS target (tag). Since mobile phones always act as initiator, that's why two phones cannot use NFC to do data exchange in such a mode.

@egislook There's indeed an API to eliminate the animation for Android beam, see here:
* https://github.com/whitedogg13/nfc-test-app

However, since it will also be deprecated in API 29, so we have no plan to bridge it currently.

amitbravo, 2021-04-02:
@whitedogg13  What about status now after a year  about "The iOS only supports NFC reader/writer mode, not p2p mode."
can exchanging data between android devices work today ?

amitbravo, 2021-04-23:
@egislook egislook did you get any luck ?

closed 2021-07-08 by github-actions[bot]