revtel/react-native-nfc-manager #259: Complete Mifare Classic Card 1k reading process

desteve, 2020-02-25:
Hi!

I'm trying to read a full Mifare Classic Card 1k modifying the example in the test project, but I can't as I'm always given the Transceive failed error on the mifareClassicReadSector.
What I'm trying is to iterate over all the sectors, authenticating and reading each one at a time and then get the data. It seems I'm capable of authenticate, but reading fails always starting on the second sector. I mean, I can read one sector, but no more.

I use this async function to read every sector:

`_readAuthenticatedA = async (sector, code) => {
return new Promise((resolve, reject) => {
      NfcManager.mifareClassicAuthenticateA(sector, code)
        .then(() => {
          console.log(`mifareClassicAuthenticateA(${sector}) completed`);
          NfcManager.mifareClassicReadSector(sector)
            .then(data => {
              console.log(`mifareClassicReadSector(${sector}) completed`);
              resolve(data);
            })
            .catch(err => {
              console.log(`mifareClassicReadSector(${sector}) failed:`, err);
              reject(err);
            });
        })
        .catch(err => {
          console.log(`mifareClassicAuthenticateA(${sector}) failed:`, err);
          reject(err);
        });
    });
  };`

Does anybody have an example of a full card reading process? Or maybe can you give me some help about why I'm getting the Transceive error?

Thank you in advance.

whitedogg13, 2020-02-26:
Hi @poison Can you help on this one?

closed 2021-05-26 by github-actions[bot]

AzranAzwer, 2023-02-09:
Anyone fix this issue ?