revtel/react-native-nfc-manager #268: "unknown error" when user cancels technology request on iOS

plol, 2020-03-09:
Hello,

When I begin a technology request and press the Cancel button in the popup i get a "unknown error" string error.

I expected to get the error code 200 as defined in https://developer.apple.com/documentation/corenfc/nfcreadererror/code/readersessioninvalidationerrorusercanceled

When looking at the code I saw that this library tends to return localised string errors which is not a great idea if i want to handle the error cases in code. I really want the error code.

Thanks

plol, 2020-03-10:
I patched it locally using patch-package with this patch:

```
% cat patches/react-native-nfc-manager+2.1.1.patch 
diff --git a/node_modules/react-native-nfc-manager/ios/NfcManager.m b/node_modules/react-native-nfc-manager/ios/NfcManager.m
index ad0d73e..40105ed 100755
--- a/node_modules/react-native-nfc-manager/ios/NfcManager.m
+++ b/node_modules/react-native-nfc-manager/ios/NfcManager.m
@@ -18,11 +18,13 @@
 NSString* getErrorMessage(NSError *error) {
     NSDictionary *userInfo = [error userInfo];
     NSError *underlyingError = [userInfo objectForKey:NSUnderlyingErrorKey];
-    NSString* result = [underlyingError localizedDescription];
-    if (result == nil) {
-        return @"unknown error";
+    if (underlyingError != nil) {
+        return [NSString stringWithFormat:@"%@:%ld,%@:%ld",
+                [error domain], (long)[error code],
+                [underlyingError domain], (long)[underlyingError code]];
     }
-    return result;
+    return [NSString stringWithFormat:@"%@:%ld",
+            [error domain], (long)[error code]];
 }
 
 @implementation NfcManager {
```

whitedogg13, 2020-03-11:
Hi @plol , that looks great! If you can make a PR for this, I will be happy to accept it and other users can definitely benefit from it.

whitedogg13, 2020-03-13:
Hi @plol , I have merged the patch and release `2.1.4`. Thanks for the contribution!

closed 2020-03-13 by whitedogg13