revtel/react-native-nfc-manager #269: How to continuously read tags?

bsbodden, 2020-03-18:
I have an application where the user needs to potentially scan a dozen or more NFC tags. If I try to put the scanning in a loop I can only successfully scan the first tag before I get `unknown error` or `you need to call registerTagEvent first`. My test looks like:

```
import React, {useEffect, useState} from 'react';
import {Button, FlatList, Text, View} from 'react-native';
import NfcManager, {Ndef, NfcTech} from 'react-native-nfc-manager';

const TestScreen = function() {
  const [tags, setTags] = useState([]);

  useEffect(() => {
    NfcManager.start();

    return cleanUp;
  });

  const cleanUp = () => {
    NfcManager.cancelTechnologyRequest().catch(() => 0);
  };

  const scanLoop = async () => {
    try {
      for (let i = 1; i < 4; i++) {
        await NfcManager.requestTechnology(NfcTech.Ndef, {
          alertMessage: `Scan tag ${i}`,
        });
        let ndef = await NfcManager.getNdefMessage();
        let text = '';
        let ndefMessage = ndef.ndefMessage[0];
        if (Ndef.isType(ndefMessage, Ndef.TNF_WELL_KNOWN, Ndef.RTD_TEXT)) {
          text = Ndef.text.decodePayload(ndefMessage.payload);
        }
        let temp = [...tags];
        temp.push(text);
        setTags(temp);
        await NfcManager.setAlertMessageIOS(`Tag => ${text}`);
        cleanUp();
      }
    } catch (ex) {
      console.warn('ex', ex);
    }
  };

  return (
    <View>
      <Button onPress={scanLoop} title="Start Scanning" />
      <FlatList
        data={tags}
        keyExtractor={index => index.toString()}
        renderItem={({item}) => <Text>{item}</Text>}
      />
    </View>
  );
};
```

I've tried variations where I do the `NfcManager.start();` and `cleanUp()` calls inside the loop. If I put the `cleanUp()` call outside the loop the alert message doesn't dismiss. I'm thinking that I might need to deal with the promises myself rather than use `await`, any hints would be appreciated. Thanks!

closed 2020-03-23 by bsbodden

hunterbmt, 2020-04-03:
@bsbodden did you find the solution for this ?

Jamjamjar, 2021-08-23:
I would also be interested to know if you resolved this?

PanosPapad, 2022-09-07:
Did the issue get resolved?

shini-tm, 2024-01-26:
I've been struggling for a while and still haven't managed to get this working yet.

```
useEffect(() => {
    // Initialize NFC Manager when the component mounts
    async function checkNfc() {
      const supported = await NfcManager.isSupported();
      if (supported) {
        console.log("TNX: NFC is supported");
        await NfcManager.start();
        setEnabled(await NfcManager.isEnabled());
        console.log("TNX: NFC is enabled");
      }
      setHasNfc(supported);
    }

    checkNfc();
  }, []);

  useEffect(() => {
    // Function to handle tag detection
    const handleTagDetection = async () => {
      try {
        //await NfcManager.start();
        console.log("TNX: NFC requestTechnology started");
        await NfcManager.requestTechnology(NfcTech.Ndef);
        console.log("TNX: NFC awaiting getTag");
        const tag = await NfcManager.getTag();
        //store tag data in state
        if (tag) {
          console.log("TNX: NFC tag found");
          const tag_id = tag.id; // Store the tag ID
          const payloadArray = tag.ndefMessage[0].payload;
          ...
      } catch (error) {
        showToast("Error: Something went wrong with NFC");
        console.error("TNX: Error initializing NFC Manager:", error?.message);
        console.log("TNX:", JSON.stringify(error));
      } finally {
        console.log("TNX: NFC cancelTechnologyRequest");
        NfcManager.setEventListener(NfcEvents.DiscoverTag, null);
        NfcManager.unregisterTagEvent().catch(() => 0); // Unregister tag event
        NfcManager.cancelTechnologyRequest();

        setNfcReset(!nfcReset); // don't need to unmount
      }
    };
    if (hasNfc && enabled) {
      handleTagDetection();
    }

    // Clean up NFC Manager when the component unmounts
    return () => {
      console.log("TNX: NFC clean up");
      NfcManager.setEventListener(NfcEvents.DiscoverTag, null); // Remove event listener
      NfcManager.cancelTechnologyRequest(); // Cancel ongoing technology request
      NfcManager.unregisterTagEvent().catch(() => 0); // Unregister tag event
    };

    // eslint-disable-next-line react-hooks/exhaustive-deps
  }, [nfcReset, setNfcReset, hasNfc, setHasNfc, enabled, setEnabled]);
```
The second read just doesn't seem to trigger the procesing functions

shini-tm, 2024-01-26:
oh lol the dealing with promises is what solved it for me:
```
useEffect(() => {
    const processTag = async (tag) => {
    ...
    };
    const handleTagDetection = async () => {
      NfcManager.registerTagEvent((tag) => console.log(tag))
        .then(() => NfcManager.requestTechnology(NfcTech.Ndef))
        .then(() => NfcManager.getTag())
        .then((tag) => {
          processTag(tag);
        })
        .then(() => setNfcReset(!nfcReset));
        
     };
     // Start listening for NFC tags when the component mounts
    if (hasNfc && enabled) {
      //handleTagDetection();
      handleTagDetection();
    }

    // Clean up NFC Manager when the component unmounts
    return () => {
      console.log("TNX: NFC clean up");
      NfcManager.setEventListener(NfcEvents.DiscoverTag, null); // Remove event listener
      NfcManager.cancelTechnologyRequest(); // Cancel ongoing technology request
      NfcManager.unregisterTagEvent().catch(() => 0); // Unregister tag event
    };
```

matth200, 2024-11-15:
I encountered a similar issue and found that the root cause was that all the functions were defined inside the useEffect block, which led to variables being scoped incorrectly and inaccessible.

To resolve this, I moved the functions outside the useEffect block. Here's how my working code looks:
```js
useEffect(() => {
  // Other setup code
}, []);

NfcManager.setEventListener(NfcEvents.DiscoverTag, (tag) => {
  console.log(tag);
  stuffToDo(tag); // Ensure this function is defined properly
});

NfcManager.setEventListener(NfcEvents.SessionClosed, () => {
  cleanUp(); // Ensure this function is defined properly
});

// Register tag event
NfcManager.registerTagEvent();
```

By restructuring this way, everything works as expected. I hope this helps others facing similar issues!