revtel/react-native-nfc-manager #271: ndefMessage payload differs on Android and iOS

chrisshiplet, 2020-03-26:
My NFC tag returns an NDEF payload of `0x2ebc5363934d591e`. On iOS, this seems to be correctly represented as the payload `[2, 101, 110, 46, 188, 83, 99, 147, 77, 89, 30]`. On Android however, I get `[2, 101, 110, 46, -68, 83, 99, -109, 77, 89, 30]`. Note the two signed numbers.

Is there any way to decode this consistently on both platforms? It seems like iOS is the correct one here.

whitedogg13, 2020-03-30:
Hi @chrisshiplet thanks for reporting this issue, I will check it out  soon.

whitedogg13, 2020-04-06:
Hi @chrisshiplet sorry for the late response! This issue has been fixed in the latest `v2.1.5`, thanks again for reporting!

closed 2020-04-06 by whitedogg13

chrisshiplet, 2020-04-06:
Awesome, thanks for the quick turnaround!

commit b805f3a6 (2020-04-06): fix #271, signed issue in Android ndef message payload