revtel/react-native-nfc-manager #272: Select PPSE Command errors (IOS)

e-fadil, 2020-03-29:
Hi, I'm not sure if this is a library issue or a limitation with IOS NFC functionality. 
Whenever I send the Select PPSE command (00a404000e325041592e5359532e444446303100) it errors with (NFCError:2)

Error Domain=NFCError Code=2 "Missing required entitlement" UserInfo={NSLocalizedDescription=Missing required entitlement}

This is the same error I get if I try to send the Select command (00A4040000)

To reproduce, 
1. Tap nfc device
2. send 00a404000e325041592e5359532e444446303100 command.

Any help would be appreciated.
Thanks!

whitedogg13, 2020-03-30:
Hi @e-fadil sorry for the late response. 

To send APDU in iOS, you will need to specify the AID you want so select in the `entitlement` file, rather then select it directly. That’s why the error message says `missing required entitlement`.

Please see [here](https://github.com/whitedogg13/react-native-nfc-manager/issues/221#issuecomment-569674952) for an example, and see Apple’s doc [here](https://developer.apple.com/documentation/corenfc/nfciso7816tag?language=objc). 

Thanks!

e-fadil, 2020-03-30:
Hi @whitedogg13 , thanks for the response! I'm assuming you can't change the selected AID after the initial nfc tap? Thanks anyway

closed 2020-03-30 by e-fadil