revtel/react-native-nfc-manager #274: Ndef record size / Multiple Ndef records?

andrerfneves, 2020-04-03:
Hi there,

Thanks for keeping this package up to date and working really well!

Trying to send a rather large string to a NFC card.

```
const bytes = Ndef.encodeMessage([
        Ndef.textRecord(
          '===lnbc10u1p0gdzxzpp5a9q65nhe9wu5y49lwtvdlr03ygy0w5cg2fkvgytswmrrrgmyhafsdp0gejk2epqgd5xjcmtv4h8xgzqypc',
        ),
        Ndef.textRecord(
          '===lnbc10u1p0gdzxzpp5a9q65nhe9wu5y49lwtvdlr03ygy0w5cg2fkvgytswmrrrgmyhafsdp0gejk2epqgd5xjcmtv4h8xgzqypc',
        ),
      ]
```

Writing some records to an NFC card and am running into issues. With 1 single `Ndef.textRecord(...)` it works. But with more than one it does not. I looked into the source code and the phonegap-nfc packages to understand limitations and while it seems a single Ndef record has a limit size (2^32 - 1 bytes?), but that I could send more than 1 record in the same `payload` (might be getting the nomenclature wrong). Is that really the case? Can I split a text string into many pieces and create many records, and then retrieve them on a separate device that scans that card?

Or what is the desired approach here? When I remove large chunks of the string, I'm able to perform the writeNdefMessage correctly.

Any hints would be appreciated! Thanks again

whitedogg13, 2020-04-06:
Hi @andrerfneves sorry for the late response. 

I think your code is correct, and the reason is most likely that your Ndef content exceeds the size limit of your NFC tag, most of them are below 200 bytes. For example, my NFC sticker uses NTAG213, the max size is about 142 bytes. 

This can be observed from native log. In the case of android, you might see something like this:
```
04-06 22:40:37.636  5109  5764 E libnfc_nci: [ERROR:nfa_rw_act.cc(1731)] Unable to write NDEF. Tag maxsize=142, request write size=220
04-06 22:40:37.636 30277 30335 D ReactNativeNfcManager: writeNdefMessage fail
04-06 22:40:37.639 30277 30334 W ReactNativeJS: writeNdefMessage fail
```

Thanks!

andrerfneves, 2020-04-08:
@whitedogg13 Thanks a lot for your input. I'm going to check on a different-sized NFC Card then. Could you point to how I'd see this `native log` exactly? And do you happen to have any resources for learning the different sizes of NFC stickers such as `ntag213` for example? Thanks!

whitedogg13, 2020-04-08:
Hi @andrerfneves for Android native log, you can try to use command like `adb logcat`.

For common tag size from NXP, you can check this [link](https://www.shopnfc.com/en/content/6-nfc-tags-specs).

andrerfneves, 2020-04-09:
Thanks a lot @whitedogg13. Appreciate the help. Closing for now!

closed 2020-04-09 by andrerfneves