revtel/react-native-nfc-manager #275: How to continuously read NFC tags

hunterbmt, 2020-04-03:
I try to implement continuously read NFC tags but the NfcManager.setEventListener only trigger event once after NfcManager.registerTagEvent, in order to receive the message again, I have to call NfcManager.unregisterTagEvent then use "magic" setTimeout to call registerTagEvent in order to prompt the NFC scan UI.
What I want to achieve is similar to this

[image]

I tried to go through the source code and document but still doesn't find out how to do it.
Please advice.

Regards,

hunterbmt, 2020-04-03:
It work on v1 but not v2, but v1 doesn't support write function.
Please advice.

whitedogg13, 2020-04-06:
Hi @hunterbmt , sorry for the late response and thanks for reporting that! 

This should be fixed in latest `v2.1.6`. You should be able to simply use `registerTagEvent` with callback to complete continuous NDEF reading like this:

```javascript
import NfcManager, {NfcEvents} from 'react-native-nfc-manager';

  _test = async () => {
    NfcManager.setEventListener(NfcEvents.DiscoverTag, tag => {
      console.warn('tag', tag);
    });

    try {
      await NfcManager.registerTagEvent();
    } catch (ex) {
      console.warn('ex', ex);
    }
  }
```

Thanks!

hunterbmt, 2020-04-06:
Dear @whitedogg13,

Just re-test on ios, I could confirm that the read and write function both working correctly on ios, will perform further tests on Android.
Thanks for your quick reply and the effort you put in for this awesome lib.

Regards,

closed 2020-04-06 by hunterbmt