revtel/react-native-nfc-manager #283: writeNdefMessage fails when its a empty tag

Edai, 2020-04-23:
Hi,

I would like to know if there is anyway to make `writeNdefMessage` works when the tag is completely empty and does not contain any Ndef? 

I have an empty tag (NFC type 5 - ISO/IEC 15693) and I would like to use writeNdefMessage to write a Ndef text record on it, but it fails on Android with 'fail to connect tag' and on iOS 'NFC Error:401; nfcd:21'.

csandman, 2020-05-20:
I am having basically the same issue.  Except in my case, if I try to write to the tags over and over again (between 3 and 25 times) eventually it will write.  I really wish I knew what was causing this.

whitedogg13, 2020-05-24:
Hi @Edai , sorry for the late response! Currently have no 15693 tags, I will do some experiments and feedback to you once I get them.

DivyaDevineni, 2021-03-22:
I am also having the same issue. Any update on this?

jungRoit, 2021-03-24:
Same Issue here as well

closed 2021-06-16 by github-actions[bot]