revtel/react-native-nfc-manager #289: Scan issues on ios

Valandur, 2020-05-12:


gbettinazzi, 2020-05-23:
Same problem here, also targetting iOS 9.0

gbettinazzi, 2020-05-23:
In my case it was my mistake because the tag I was using was empty

Valandur, 2020-05-23:
Considering it works on Android I don't think an empty tag is the issue, at least for me.

whitedogg13, 2020-05-24:
Hi @Valandur , I think @gbettinazzi is right. Tags contain no valid NDEF records will not be discovered by `registerTagEvent`, since we use `NFCNDEFReaderSession` under the hood for this case.

Can you try it again with tags contain valid NDEF message?

Valandur, 2020-05-25:
Hi, sorry for the confusion, but I am actually using a tag with data. I was just thinking that the fact that it has data shouldn't matter for the firing of the event, but it possibly does?

The tags I'm reading do contain a valid NDEF message, and I also receive that data on Android. On iOS I only receive the data when adding the fix I posted above.

Valandur, 2020-05-27:
I believe to have found the cause for this issue. I had to set `platform :ios, '11.0'` in my `Podfile`, not just in the main project. Now it seems to work, even without the additional line changes I mentioned.

closed 2021-05-26 by github-actions[bot]