revtel/react-native-nfc-manager #291: error on writing Ndef on iOS 13.4.1

Popl7, 2020-05-17:
thx for the great library. I am having great fun with these tags :-)

I have made a small test program to read and write NTAG213 on both Android and iOS.
Reading works perfectly but I get an error on my iPhone when writing.
This does work on my Android phone.

I am sort of stuck here.
Maybe I am missing something in my setup?

I am using the latest iPhone: 2020 SE with iOS 13.4.1.
In Xcode and on the Apple Developer Program I have added NFC entitlements.
When I write with the iPhone I get an error: 'NFCError:2'

This is the code I use for writing:
```
      await NfcManager.requestTechnology(NfcTech.Ndef);
      await NfcManager.getNdefMessage();

      let bytes = buildUrlPayload(urlToWrite);
      await NfcManager.writeNdefMessage(bytes);
```

The error is thrown on the NfcManager.requestTechnology call.

Is there something I forgot?

faizalshap, 2020-05-20:
Hi @Popl7 
i had the same issue it turns out you need to add few things in info.plist
`<key>com.apple.developer.nfc.readersession.felica.systemcodes</key>
	<array>
		<string>12FC</string>
	</array>
	<key>com.apple.developer.nfc.readersession.formats</key>
	<array>
		<string>NDEF</string>
		<string>TAG</string>
	</array>
	<key>com.apple.developer.nfc.readersession.iso7816.select-identifiers</key>
	<array>
		<string>D2760000850101</string>
	</array>`

Also, remove nfc reading capabilities and add again.

Popl7, 2020-05-20:
thx @faizalshap 

this was exactly what I needed.
I did read about these codes but I could not find an example.
This worked for me perfectly.

closed 2020-05-20 by Popl7

daheeahn, 2020-07-20:
I

> Hi @Popl7
> i had the same issue it turns out you need to add few things in info.plist
> `<key>com.apple.developer.nfc.readersession.felica.systemcodes</key> <array> <string>12FC</string> </array> [...]

It doesn't work. any solution?