revtel/react-native-nfc-manager #293: How to add a password to a tag

traxx10, 2020-05-21:
I want to add a feature to my app where i make tags that are password protected and only my app can remove the password and enable writing on the tag.

There is no feature yet to enable password protection on tags but on mobile applications like **NFC TOOL**, there are features to add a password and also to remove a password.

whitedogg13, 2020-05-24:
Hi @traxx10 indeed, we have no feature about adding / removing a password. This is because normally the password protection feature is tag specific, and can be implemented by low-level NFC operation, which is the actual focus of this library.

What tag do you use? We can try to add some examples for demonstration.

traxx10, 2020-05-24:
Thanks for the speedy reply.
I would appreciate if you can provide an example.
I use **Mifare Classic Tag**

I would await your feedback.
Regards Ben.

gbettinazzi, 2020-09-28:
Any update on this? We are using password-protected Mifare Ultralight tags and we want to be able to re-write the tags with a react native application. 
@whitedogg13

whitedogg13, 2020-09-28:
Hi @traxx10 sorry, we don't support MifareClassic

@gbettinazzi for Mifare Ultralight, it should work with following code snippet:

```javascript
let resp;
const password = [0x1, 0x2, 0x3, 0x4];

if (Platform.OS === 'ios') {
  await NfcManager.requestTechnology([NfcTech.MifareIOS]);
  resp = await NfcManager.sendMifareCommandIOS([0x1b, ...password]);
} else {
  await NfcManager.requestTechnology([NfcTech.Ndef]);
  resp = await NfcManager.transceive([0x1b, ...password]);
}
```

gbettinazzi, 2020-09-28:
@whitedogg13 thanks for the replay, let me get this straight. This snippet is to actually set a password to a Tag right? 

My example is slightly different: I already have a password protected tag and I want to re-write it (knowing the password), so I guess the flow should be:

* Remove the password protection
* Write new Ndef records
* Set the password again

I'm I correct? Any suggestion on wow can I implement this flow with your (beautiful) library? (:

MaoJJJJJ, 2021-04-07:
> Hi @traxx10 sorry, we don't support MifareClassic
> 
> @gbettinazzi for Mifare Ultralight, it should work with following code snippet:
> 
> ```js
> let resp;
> const password = [0x1, 0x2, 0x3, 0x4];
> 
> if (Platform.OS === 'ios') {
>   await NfcManager.requestTechnology([NfcTech.MifareIOS]);
>   resp = await NfcManager.sendMifareCommandIOS([0x1b, ...password]);
> } else {
>   await NfcManager.requestTechnology([NfcTech.Ndef]);
>   resp = await NfcManager.transceive([0x1b, ...password]);
> }
> ```

Seems like this example is only work with IOS, is there a possible way for Android to do the same thing? Thank you. @whitedogg13

MaoJJJJJ, 2021-04-08:
> Hi @traxx10 sorry, we don't support MifareClassic
>
> @gbettinazzi for Mifare Ultralight, it should work with following code snippet:
>
> ```js
> let resp;
> const password = [0x1, 0x2, 0x3, 0x4];
>
> if [...]

@whitedogg13 Hello, thank you for this nice library. 

I am currently having trouble setting password using only Ndef service as it is actually not supported in the android code(transceive function in NfcManager), so I have to cancel the Ndef service and request MifareUltralight service separately. Is there a way to write and set password to the tag just using one tag event?

Cheers.

MaoJJJJJ, 2021-04-13:
If anyone still wanna add password for Android please check this page:
https://stackoverflow.com/questions/22719465/ntag212-mifare-ultralight-with-authentication/22723250#22723250

@gbettinazzi for removing the password protection, authorise the tag first according to the page above and set the AUTH0 to 0x0ff, like below, I am using NTAG213 so it should be on page 41:
```
const result = await NfcManager.transceive([
          0x30, // READ
          0x29, // PAGE 41
]);

if (result != null && result.length >= 16) {
          // read always returns 4 pages
          const response = NfcManager.transceive([
            0xa2, // WRITE
            0x29, // PAGE 41
            result[0], // keep old value for byte 0
            result[1], // keep old value for byte 1
            result[2], // keep old value for byte 2
            0x0ff, // disable protection
]);
```

Then you can reset the password and pk as you want following the instructions.

traxx10, 2021-04-13:
> If anyone still wanna add password for Android please check this page:
> https://stackoverflow.com/questions/22719465/ntag212-mifare-ultralight-with-authentication/22723250#22723250
>
> @gbettinazzi for [...]

Thanks for this, i'll test and see

farry19, 2021-06-28:
Can some please guide me how to write custom tags e.g. vCard as there is no docs available and i am trying to search for tutorials and examples that how and what is the difference between `NDef` and `Mifare`

traxx10, 2021-08-29:
Resolved

closed 2021-08-29 by traxx10

ThirupathiRaoM, 2022-01-25:
Hello Everyone 
I am working NFC Application in swift5 I am trying to password protected using Core NFC if any body have idea how to password protected the NFC Tag using coreNFC Please let us know it will help me a lot

AnirudhGudugunti, 2024-06-20:
> Closing the issue

Hey @traxx10 did u able to set and remove the password for NFC tags. As i am trying to set the password for NFC tags(NTAG21x series NTAG213,NTAG216 MifareUltralight) in android, i am getting "transceive fail" error everytime. Below i have pasted my code to set the password for tag please look in to it .                                                                                               
  const writePasswordToTag = async () => {
    const defaultPack = [0x12, 0x34]; // Default PACK value
    try {
      await NfcManager.requestTechnology(NfcTech.NfcA);
      const tag = await NfcManager.getTag();
      const idBytes = tag.id.split(/(..)/g).filter(s => s).map(val => parseInt(val, 16));
      const pwdBytes = [
        parseInt(password.substring(0, 2), 16),
        parseInt(password.substring(2, 4), 16),
        0x00,
        0x00,
      ];
      const commands = [
        [0xA2, 0x2B, ...pwdBytes], 
        [0xA2, 0x2C, defaultPack[0], defaultPack[1]], // Write default PACK
        [0xA2, 0x29, 0x04], // Set AUTH0 to protect starting from page 4
      ];
      for (const command of commands) {
        const response = await NfcManager.nfcAHandler.transceive(command);
        //const response = await NfcManager.nfcAHandler.transceive(command)
        if (!response) {
          throw new Error('transceive fail');
        }
      }
      Alert.alert('Success', 'Password set successfully.');
    } catch (ex) {
      console.warn('Error during password set:', ex);
      Alert.alert('Error', 'Failed to set password: ' + ex.message);
    } finally {
      NfcManager.cancelTechnologyRequest();
    }
  };
And if u have done this please provide your code. Please respond to this ASAP.
Thanks.