revtel/react-native-nfc-manager #295: NFCError:2 when calling await NfcManager.requestTechnology on iOS

vince0656, 2020-05-22:
Hi there,

I'm attempting to read the ID of a mifare NFC chip on iOS. I can read the text layer OK when using:

    await NfcManager.registerTagEvent();

i.e. the system dialog pops, the data can be parsed fine etc. However, in the same app, when I attempt to run the following (in place of `registerTagEvent`):

    await NfcManager.requestTechnology([
        NfcTech.MifareIOS,
        NfcTech.Iso15693IOS,
        NfcTech.IsoDep,
      ]);

I get the following errors:

    NFCTag didInvalidateWithError
    [tid:com.facebook.react.JavaScript] 'ex', 'NFCError:2'

This is before I even call `getTag()`. 

Any ideas on where I'm going wrong?

`Info.plist` has the privacy description, my provisioning profile allows NFC reading and my entitlements file looks like this:

```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>com.apple.developer.nfc.readersession.formats</key>
	<array>
		<string>NDEF</string>
		<string>TAG</string>
	</array>
</dict>
</plist>
```

I am getting the following message after calling NFCManager.start():
```
NfcManager initialized
```

So that seems to be a good sign. 

Further info: 
 - Running version `2.1.8` of the manager
 - App running on iOS 13.4.1

vince0656, 2020-05-22:
Just got it working on version `2.1.5` - so I guess this is definitely a bug in the latest version

vince0656, 2020-05-22:
Will keep bumping the version until I find the one that breaks

whitedogg13, 2020-05-24:
Hi @vince0656 thanks for reporting this! 

I can reproduce the some condition as you, and the root cause seems to relate to the latest iOS feature targeting `Felica`. I found that once the `NFCPollingISO18092` options is used when initializing `NFCTagReaderSession`, this error will pop up.

Just publish `v2.1.9` as a temp fix for this issue by disabling `NFCPollingISO18092`, please give it a try!

vince0656, 2020-05-28:
@whitedogg13 Thanks so much `2.1.9` is working as expected

closed 2021-05-26 by github-actions[bot]