revtel/react-native-nfc-manager #298: NFC Error:403 on new NFC chips

jackgisel, 2020-06-09:
I feel like I am having a simple issue, but I can figure it out! I am getting a 403 error on trying to write to new NFC chips.

V2.1.9 

ps: thanks so much for your long support of this package

jackgisel, 2020-06-13:
Was able to fix this. My issue was dealing the NDEF. I had to use custom commands.

closed 2020-06-13 by jackgisel

crobinson42, 2020-10-05:
@jackgisel how did you solve this with custom commands? Do you have a snippet to share?

whitedogg13, 2021-02-10:
Some updates. I have made a naive format example, please check here: https://github.com/revtel/react-native-nfc-rewriter/tree/naive-ndef-format

However, this is actually a naive approach, because we omit some important control message (ex: tag lock information) before actual NDEF TLV.

Hope that helps!