revtel/react-native-nfc-manager #299: [iOS] RCTBridgeModule.h: React/RCTDefines.h file not found

rahamin1, 2020-06-12:
Trying to build on iOS (RN 0.62.2, react-native-nfc-manager 2.1.9), the build aborts with the following error (I am copying the text and the screenshot of the same text, viewing it is more convenient):

```
/react-native/React/Base/RCTBridgeModule.h:10:9: 'React/RCTDefines.h' file not found
/Users/user172615/Documents/rn62-take2/plumpclient/node_modules/react-native-nfc-manager/ios/NfcManager.m:1:9: In file included from /Users/user172615/Documents/rn62-take2/plumpclient/node_modules/react-native-nfc-manager/ios/NfcManager.m:1:
/Users/user172615/Documents/rn62-take2/plumpclient/node_modules/react-native-nfc-manager/ios/NfcManager.h:7:9: In file included from /Users/user172615/Documents/rn62-take2/plumpclient/node_modules/react-native-nfc-manager/ios/NfcManager.h:7:
```

[image]

whitedogg13, 2020-06-24:
@rahamin1 Thanks for reporting, maybe something related to RN62? Currently my other project use RN 61.5 and it works propertly. I will test this later.

rahamin1, 2020-06-25:
Thanks, but the problem doesn't exist if you start a new rn project in 0.62.2.
It happened when I upgraded a project from 0.59.9.

I am closing this issue, since the fault is probably not in this package.

closed 2020-06-25 by rahamin1