revtel/react-native-nfc-manager #300: General Question -- Reliability across iPhone versions

bsbodden, 2020-06-12:
I have an application on the field, deployed across iPhone 7 to 11. We are using:

- ISO 14443-3A - Type A NXP Mifare Ultralight - NTAG213

We are getting reports of failed scans on iPhone 7 in particular, we get the on > 8 but they are sporadic.

The bulk of our scanning code looks like:

```
export function initializeNfc(callback = function() {}) {
  NfcManager.start()
    .then(callback())
    .catch(error => {
      let message =
        error === 'Not support in this device'
          ? 'NFC is not supported on this device'
          : error;
...
    });

  return nfcCleanUp;
}

export function nfcCleanUp() {
  NfcManager.cancelTechnologyRequest().catch(() => 0);
}
```

On our screens we use it as follows:

```
  useEffect(() => {
    return initializeNfc();
  }, []);
```

Our *scanning* needs to either grab the NDEF text record or the UID of the tag, it look like:

```
export async function nfcScan(prompt, successMessage) {
  initializeNfc();
  let isEmulator = await DeviceInfo.isEmulator();
  if (!isEmulator) {
    try {
      await NfcManager.requestTechnology(NfcTech.Ndef, {
        alertMessage: prompt,
      });
      let ndef = await NfcManager.getNdefMessage();
      const textPayload = getNdefText(ndef);
      await NfcManager.setAlertMessageIOS(successMessage);

      return {success: true, type: 'ndef', payload: textPayload};
    } catch (ex) {
      console.log('nfcScan: ex ==>', JSON.stringify(ex));
      if (ex === 'NFCError:403') {
        // Tag doesn't contain an NDEF message
        // Some old KleenEdge Tags won't have one and the NFC uid has to be used
        let nfcUid = await NfcManager.getTag();
        await NfcManager.setAlertMessageIOS(successMessage);

        if (typeof nfcUid === 'string') {
          return {success: true, type: 'uid', payload: nfcUid};
        } else if (typeof nfcUid === 'object') {
          if (Object.entries(nfcUid).length === 0 || nfcUid.id == null) {
            return {success: false, error: 'Invalid or Damaged Tag'};
          } else {
            return {success: true, type: 'uid', payload: nfcUid.id};
          }
        }
```

In our screen we typically have a method like:

```
  const scan = async () => {
    try {
      const response = await nfcScan(
        'Hold phone near a Tag to scan',
        'Tag read successfully',
      );

      ...
```

We tried to catch all the exceptions we know about (this seems to be very poorly documented). For that we do the following:

```
 } catch (ex) {
      if (ex === 'NFCError:403') {
        // Tag doesn't contain an NDEF message
        // Some old KleenEdge Tags won't have one and the NFC uid has to be used
        let nfcUid = await NfcManager.getTag();
        await NfcManager.setAlertMessageIOS(successMessage);

        if (typeof nfcUid === 'string') {
          return {success: true, type: 'uid', payload: nfcUid};
        } else if (typeof nfcUid === 'object') {
          if (Object.entries(nfcUid).length === 0 || nfcUid.id == null) {
            return {success: false, error: 'Invalid or Damaged Tag'};
          } else {
            return {success: true, type: 'uid', payload: nfcUid.id};
          }
        }
      } else if (ex === 'NFCError:200') {
        return {
          success: false,
          type: 'cancel',
          error: 'Scan was cancelled',
        };
      } else if (
        //
        // Unrecoverable NFC Errors
        //
        ex === 'unknown error' ||
        ex === 'NFCError:201' ||
        ex === 'NFCError:202' ||
        ex === 'NFCError:203' ||
        ex === 'nfcd:37' ||
        ex === 'nfcd:28'
      ) {
        return {
          success: false,
          type: 'error',
          error: 'Error reading the Tag, Please try again',
        };
      } else if (ex === 'NFCError:204') {
        // these are bad tag or irrecoverable errors
        return {
          success: false,
          type: 'error',
          error: 'Invalid or Damaged Tag',
        };
      } else {
        return {
          success: false,
          type: 'error',
          error: 'Error reading the Tag',
        };
      }
    } finally {
      nfcCleanUp();
    }
```
What a looking for is some guidance if we are doing anything that could possible be causing failed scans, or if this is just the nature of NFC on older devices like the iPhone 7.

rahamin1, 2020-07-01:
@bsbodden Any news on that?

bsbodden, 2020-07-03:
> @bsbodden Any news on that?

No, but I can confirm that when they encounter a failed scan, retrying the scan does the trick and failures are more prevalent the older the device is.

closed 2021-05-26 by github-actions[bot]