revtel/react-native-nfc-manager #304: Ndef technology no longer available after error during writeNdefMessage

faltakam, 2020-06-19:
Hello,

i was writing Ndef messages to tag and at first it worked fine. But after a while i got error during writeNdefMessage. After that i was not able to read or write Ndef message. I found out that the Ndef technology was no longer available on the tag.

Here is tag before the error:
`{"canMakeReadOnly": false, "id": "028500054575ED", "isWritable": true, "maxSize": 2046, "ndefMessage": [{"id": "", "payload": [Array], "tnf": 1, "type": [Array]}], "techTypes": ["android.nfc.tech.IsoDep", "android.nfc.tech.NfcA", "android.nfc.tech.Ndef"], "type": "NFC Forum Type 4"}`

Here is tag after the error:
`{"id": "028500054575ED", "techTypes": ["android.nfc.tech.IsoDep", "android.nfc.tech.NfcA"]}`
Here 

Here is my code i used to write messages:
```
const buildPayload = (message: string) => {
  return Array.from(Ndef.encodeMessage([Ndef.textRecord(message)]));
};

export const writeMessage = async (message: string) => {
  try {
    console.log('[NFC] writing message ', message);
    await NFC.requestTechnology(NfcTech.Ndef);
    const tag = await NFC.getTag();
    console.log('[NFC] tag ', tag);
    const payload = buildPayload(message);
    console.log('[NFC] writing message ', payload);
    await NFC.writeNdefMessage(payload);
    console.log('[NFC] message written');
    await cleanUp();
  } catch (e) {
    console.warn('[NFC] error writing:', e);
    await cleanUp();
    throw e;
  }
};
```

Is there a way to restore the tag so Ndef technology is available again and i can read/write Ndef messages?

faltakam, 2020-06-22:
I tried another device and for that device Ndef technology was available (the Ndef tag was empty). After i wrote data into the tag i was able to read and write Ndef tag message from original device again.

closed 2020-06-22 by faltakam

cjbreg, 2021-03-12:
I have the same problem, did you find a way to prevent this from happening in the future?

faltakam, 2021-03-12:
I did (kind of). In my use case the tag was used to communicated with module that was connected to tag using I2C interface. So i have asked the guys who programmed the module to periodically (every 30s) write some data to the tag if the tag is empty. This was good enough solution for my use case.