revtel/react-native-nfc-manager #305: Test suite failed to run. Invariant Violation: Native module cannot be null.

DanielDanaee, 2020-06-23:
RN 0.61.2
react-native-nfc-manager: ^2.1.4

[image]

whitedogg13, 2021-01-21:
Hi @alexcastillo 

I think you will need to mock the native module, since it doesn't exist in node environment. 

Something like this: https://github.com/naoufal/react-native-safari-view/issues/99

Thanks!

alexcastillo, 2021-01-21:
Thanks for replying, @whitedogg13.

I should have clarified. I'm not trying to run tests, I'm just running the app.

I guess my issue has the same error. Please refer to #193 #148 #93

alexcastillo, 2021-01-21:
Here's the output:

[image]

whitedogg13, 2021-01-21:
@alexcastillo @DanielDanaee For test fail, please try my suggestion above. I will close this issue for now, please refer to #193 for the same condition when actually running application.

closed 2021-01-21 by whitedogg13