revtel/react-native-nfc-manager #306: requestTechnology not working on android signed app without debug enabled

JonathanMM, 2020-06-24:
Hello everyone,
I create an application to read public transport card of Paris' region (the passe Navigo). It works when I create a debug or release version of app, but when I create a release signed version (for playstore), the application doesn't work. More precise, if I create a signed version in release with android:debuggable="true" in manifest, it works, if I create a signed version and I start on my phone with a USB cable plugged (with adb and react-native server running), it works, but if I start the application without USB cable, it doesn't work, for me and everyone who downloaded my application on play store.
When I say it doesn't work, the app starts, the user click on "Lire un passe" (read a card in english), which call 
```typescript
await NfcManager.requestTechnology(NfcTech.IsoDep);
```
the card is detected by the phone (a sound played) but nothing happens. Just after this line, I have a setState which change the screen displayed to the user (to say don't move the card), and it newer displayed in this case.

Source code: https://framagit.org/JonathanMM/navigoreader/-/blob/master/App.tsx#L41

Have you any idea to solve my issue?

JonathanMM, 2020-06-24:
More information: it works in android 9, the issue seems happen only in android 10

jackgisel, 2020-09-17:
It's not working either for me in Android v10. The call doesn't respond any bugs or issues which is very odd.

JonathanMM, 2021-07-12:
Still happen

closed 2021-09-28 by github-actions[bot]