revtel/react-native-nfc-manager #313: [IOS] Present-sector password from ISO15693 tags - Tag connection lost

Johbrun, 2020-07-21:
Hey ! 
Im trying to send a command to unlock tag. I used B3h command.

But it doesn't work, I received "Tag connection lost"

Im using a m24lr64-r tag with64k EEPROM (and so I have to using extended_flag for reading, but its an other problem)

The command work fine on Android (with transceive())

Here my code : 

```
// not addressed mode
const data = [0x02, 0xb3, 0x02, 0x01, passByte[0], passByte[1], passByte[], passByte[3] ];

const flags = bytes[0]; // 0x02 : High Data Rate. No need to configure extended flag
const customCommandCode = bytes[1];
const customRequestParameters = bytes.slice(2, bytes.length);

resp = await handler.customCommand({
      flags,
      customCommandCode,
      customRequestParameters
    })
```

Morever, I saw that extended flag was no available on ios13 : https://community.st.com/s/question/0D50X0000B09Q8SSQU/example-of-using-the-new-ios-13-corenfc

Do you know if it is still the case ?

Johbrun, 2020-11-13:
Always same problem with iOS14 :/

closed 2021-05-26 by github-actions[bot]

zabojad, 2021-08-12:
@Johbrun I'm having a similar issue with an ISO15693 tag. Reading and writing memory is protected with a password.

I've tried `handler.customCommand()` as you use it but it fails (empty error with no message)...

Have you managed to make it work on iOS ?

zabojad, 2021-08-12:
@Johbrun my bad, I'm getting the exact same error as you actually:

```
[CoreNFC] 00000002 81ac0240 -[NFCTagReaderSession transceive:tagUpdate:error:]:789  Error Domain=NFCError Code=100 "Tag connection lost" UserInfo={NSLocalizedDescription=Tag connection lost}
```

Did you solve it?

zabojad, 2021-08-12:
I found it!

When you look at the iOS documentation about [`customCommandWithRequestFlag`](https://developer.apple.com/documentation/corenfc/nfciso15693tag/3043799-customcommandwithrequestflag?language=objc), you can read this little note at the bottom of that page:

> This method inserts the icManufacturerCode after the command byte before appending the customRequestParameters when constructing the data packet that the method sends to the tag.

Gosh, how obvious was it that they insert it by default?! <strike>Damn you Apple!</strike>

I removed the `0x02` at the beginning of the parameters array and it works.

In your case, you  should change this:
```
const customRequestParameters = bytes.slice(2, bytes.length);
```
to this:
```
const customRequestParameters = bytes.slice(3, bytes.length);
```

It's probably too late now to give you this answer but anyway, that might help other folks out here...

Johbrun, 2022-06-22:
Hello @zabojad ,

WOOOOW thank you very much, you saved my day ! 
Thank you for your response !

For information I use now an other tag (ST25DV64K)

zabojad, 2022-06-24:
@Johbrun I see you are in Nantes, we are quite close geographically. Feel free to connect with me on LinkedIn or some other platform...

danieliftikhar41, 2022-10-05:
Hello!

I had a problem sending the 0xB3 custom command using the handler.customCommand function. 

Here is my code, I can't see any error there, but my RN code only will give me that [error] message, without giving me any information about what is going on.

```
const data=[0x02,0xb3,0x02,0x01,pass[0],pass[1],pass[2],pass[3],pass[4],pass[5],pass[6],pass[7]];
const flags = data[0];
      const customCommandCode = data[1];
      const customRequestParameters = data.slice[3,data.length]
try {
        resp = await handler.customCommand({
          flags,
          customCommandCode,
          customRequestParameters        
        });
```

Giving me this error:
`Error [error]`
Could you help me out?

Thanks!

ramazanerdogan, 2023-07-24:
I'm also getting the same error, can anyone help with a solution?

ramazanerdogan, 2023-07-24:
@Johbrun  can you help me if you solved this situation please ?