revtel/react-native-nfc-manager #317: Make App as NFC Card

Ov3rControl, 2020-08-05:
Hello we are building smart doors that people can put their NFC cards on to unlock the door.
wondering if i can do the same thing but on mobile with this library? create an app that acs as a NFC Card (Not Reading or Writing)
The typical senario is i added some string on the mobile NFC when i put the app in front of the door. the door will scan the NFC data on the app and open the door.

How can i do something like that?

StorkZ, 2020-08-18:
https://stackoverflow.com/questions/34607168/nfc-unavailable-when-camera-is-open

jackgisel, 2020-09-13:
That would be really interesting. But, Apple doesn't give permission to do this. It's a closed API for their wallet app.

svvitale, 2020-09-14:
This can be done on Android using card emulation, but as @jackgisel mentioned, not possible on iOS.

appidea, 2020-09-27:
Hi, as @svvitale mentioned, it is possible to do using the HCE feature in Android. I just wrote the react-native extension "[react-native-hce](https://github.com/appidea/react-native-hce)", that supports the NFC tag emulation right away. Hope it will help.

amitbravo, 2021-03-25:
@appidea so we can transmit from data from device A using react-native-hce and can receive data on B devices using react-native-nfc-manager , right ?

amitbravo, 2021-04-02:
@Ov3rControl did you find any solution ?

Ov3rControl, 2021-04-02:
@amitbravo no the project was cancelled so i didn't get the chance

amitbravo, 2021-04-02:
can't simply 
react-native-nfc-manager work as sending data on one device and read on another device , same app .

closed 2021-06-18 by github-actions[bot]

Malthehave, 2021-06-20:
@amitbravo did you ever figure out a solution for this? Thanks

jeffinhk, 2023-06-05:
any updates on this one?