revtel/react-native-nfc-manager #319: Doesn't seem to work with IOS 14 Beta

TylerNRobertson, 2020-08-14:
Im not expecting this library to work with pre-release beta software but I'm putting this here because I have been experimenting & ran into this issue once the device is on IOS 14 Beta 4.

Inside the JS debugger you get: "No active session"

Inside XCode you receive: [30225:2089137] [CoreNFC] 00000002 8181d600 -[NFCTagReaderSession beginSessionWithConfig:]:315  error:Error Domain=NFCError Code=203 "System resource unavailable" UserInfo={NSLocalizedDescription=System resource unavailable}, errorCode: 0xcb

TylerNRobertson, 2020-08-24:
@laurensk i just updated mine, mustve been an IOS specific issue.

hunterbmt, 2020-08-25:
It's IOS specific issue, if I hard reset the iPhone, NFC Reader work for 1 time then not working anymore.

laurensk, 2020-08-25:
Are you on Beta 5 or not?

laurensk, 2020-08-26:
Summary: It seems like iOS 14 Beta 4 (and older) had issues with NFC scanning in general and it apparently has nothing to do with this package at all. Closing this issue now! Thanks for participating.

jackgisel, 2020-09-17:
This can be closed.

closed 2020-09-17 by TylerNRobertson