revtel/react-native-nfc-manager #323: General Question : How to lock a tag ?

SlyBcDev, 2020-09-04:
Hi,

 I use this library and it's work fine, really thanks for your job!

 So I saw that some app can definitivly  lock a tag to be writing. I would like to know if there is a possibility to do this with react-native-nfc-manager ?

Thanks
Sylvain

jackgisel, 2020-09-13:
I think it is possible through the custom commands. I might work on that feature and create a PR at add it.

traxx10, 2020-09-15:
> I think it is possible through the custom commands. I might work on that feature and create a PR at add it.

That would be great.
I would need this feature, please.
You can reach me at codetraxx10@gmail.com

thatavivku90, 2020-09-19:
hey i would love to get more info about it too :) thanks!
avivku90@gmail.com

mehdinasiripoor, 2020-11-01:
@jackgisel I would need this, thanks!
mehdi.nasiripoor@yahoo.com

traxx10, 2021-01-18:
@jackgisel any updates on this?

closed 2021-05-26 by github-actions[bot]