revtel/react-native-nfc-manager #326: Xcode 12 & iOS 14 run error..

TrustDec, 2020-09-17:
```
dyld: Library not loaded: /usr/lib/libnfshared.dylib
  Referenced from: /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Library/Developer/CoreSimulator/Profiles/Runtimes/iOS.simruntime/Contents/Resources/RuntimeRoot/System/Library/Frameworks/CoreNFC.framework/CoreNFC
  Reason: image not found
dyld: launch, loading dependent libraries
DYLD_SHARED_CACHE_DIR=/Users/XXX/Library/Developer/CoreSimulator/Caches/dyld/20A5364e/com.apple.CoreSimulator.SimRuntime.iOS-14-0.18A372
DYLD_ROOT_PATH=/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Library/Developer/CoreSimulator/Profiles/Runtimes/iOS.simruntime/Contents/Resources/RuntimeRoot
DYLD_LIBRARY_PATH=/Users/trust/Library/Developer/Xcode/DerivedData/MobileIntegration-eyrbnwcfymemnmgrsfzearpkmypp/Build/Products/Debug-iphonesimulator:/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Library/Developer/CoreSimulator/Profiles/Runtimes/iOS.simruntime/Contents/Resources/RuntimeRoot/usr/lib/system/introspection
DYLD_INSERT_LIBRARIES=/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Library/Developer/CoreSimulator/Profiles/Runtimes/iOS.simruntime/Contents/Resources/RuntimeRoot/usr/lib/libBacktraceRecording.dylib:/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Library/Developer
```

toshe, 2020-09-17:
This seems to be related to this thread and CoreNFC in general:

https://stackoverflow.com/questions/63915728/xcode12-corenfc-simulator-library-not-loaded

The simulator doesn't support NFC but the libraries are included when compiling. Since CoreNFC is not compiled for sim it won't work. I would start with inspecting all the NFC references and exclude them when building for simulator.

toshe, 2020-09-17:
@TrustDec does this happen only in a simulator or also on a real device?

toshe, 2020-09-17:
https://developer.apple.com/forums/thread/660398 here's more info related to the thread.

whitedogg13, 2020-09-20:
@TrustDec have you ever tried RN62? My app works fine with RN62.2. I will try to use RN63 and feedback later.

toshe, 2020-09-20:
I'm using RN62 and can confirm that this issue happens in a simulator on iOS 14. The links above show more info.

whitedogg13, 2020-09-20:
@toshe Thanks! you're totally right and I can also reproduce this. It looks like an issue from Apple, I will update README to warn other users. Thanks again for bringing this.

ibogdan94, 2020-09-28:
I'm using RN 63.2 and xcode Version 12.0.1 (12A7300) -- the problem still happens on all simulators on iOS 14

sallar, 2020-09-30:
The funny thing is it's not possible to download and install iOS 13 Simulator because the list of runtimes is empty https://developer.apple.com/forums/thread/662072

tsiory, 2020-10-02:
It's only happens on simulator or device too? Thanks!

toshe, 2020-10-02:
The issue affects ONLY iOS 14 simulators in XCode 12.0 and 12.0.1.
It doesn't affect devices (iOS 13.1.1 and iOS 14.0.1).

toshe, 2020-10-02:
@sallar in my XCode I have the options of iOS 14 and iOS 13.5 simulators. The iOS 13.5 simulators work without issues.

sallar, 2020-10-02:
@toshe yeah that particular issue is only on 12.0.1. 12.0 is fine and i can download 13.5 there

closed 2021-05-26 by github-actions[bot]

ashok-sl, 2021-11-17:
IOS 15, It is still crashing in simulator, however, it works on real devices.
```
xCode: 13.1
SDKs:
    iOS SDK:   Platforms: DriverKit 21.0.1, iOS 15.0, macOS 12.0, tvOS 15.0, watchOS 8.0
npmPackages:
    @react-native-community/cli: ^6.1.0 => 6.1.0 
    react: ^17.0.2 => 17.0.2 
    react-native: 0.66.1 => 0.66.1 
```