revtel/react-native-nfc-manager #331: Write and Read MiFare Classic

BravenxX, 2020-09-30:
I have this functions that read the message in a miFareClassic then write a new message and returns the chip uid and the message readed from the chip.  

```
import NfcManager, { NfcTech, Ndef } from 'react-native-nfc-manager';

const writeAndReadMiFare  = (message) => {

  const resp = await NfcManager.requestTechnology(NfcTech.MifareClassic);
  const tag = NfcManager.getTag();
  console.warn('tag', tag); // => { "_40":0 , "_65":0, "_55":null, "_72":null } 

  const encoded = message // => How can I encode this string?
  const wrote = await NfcManager.mifareClassicWriteBlock(0, encodedMessage);

  const readed = await NfcManager.mifareClassicReadBlock(0);
  console.warn('readed', readed); // => [178, 0,0,225,16, ...]

  return { uid: tag?.id, message: readed  }; 
}
```

But in miFareClassic I can't make this work, you can see what I get when I scan a chip in comments. I'm new on this and I can't find how to do this on the docs, How do I encode y de-encode messages and how can I get the uid?. I think is very simple but I can't find the methods to do it work for miFareclassic.

BravenxX, 2020-10-15:
For those who need it, this is how I did it with react-native-nfc-manager V2.0 to read and write in MiFareClassic Card

```
import React, { useState, useEffect } from 'react';
import {
  View,
  Text,
  Platform,
  TouchableOpacity,
  ScrollView,
} from 'react-native';
import NfcManager, {
  ByteParser,
  NfcTech,
  NfcEvents,
} from 'react-native-nfc-manager';

const KeyTypes = ['A', 'B'];

const KEY = KeyTypes[1];
const KEY_TO_USE = 'FFFFFFFFFFFF';
const SECTOR_TO_WRITE = 1;
const TEXT_TO_WRITE = 'Hello World!';

const App = () => {
  const [nfcProps, setNfcProps] = useState({
    supported: false,
    enabled: false,
    isDetecting: false,
    mode: 'read',
    tag: null,
    sectorCount: null,
    blocksInSector: null,
    parsedText: null,
    firstBlockInSector: null,
  });

  useEffect(() => {
    NfcManager.isSupported(NfcTech.MifareClassic).then((supported) => {
      if (supported) {
        _startNfc();
        NfcManager.start()
          .then(() => NfcManager.isEnabled())
          .then((enabled) =>
            setNfcProps({ ...nfcProps, enabled, supported: true }),
          )
          .catch((err) => {
            console.warn(err);
            setNfcProps({ ...nfcProps, enabled: false });
          });
        NfcManager.setEventListener(NfcEvents.DiscoverTag, (tag) => {
          console.warn('tag', tag);
          NfcManager.setAlertMessageIOS('I got your tag!');
          NfcManager.unregisterTagEvent().catch(() => 0);
        });
      }
    });

    return () => {
      NfcManager.cancelTechnologyRequest().catch(() => 0);
      NfcManager.setEventListener(NfcEvents.DiscoverTag, null);
      NfcManager.unregisterTagEvent().catch(() => 0);
    };
  }, []);

  const _startDetection = async () => {
    const cleanUp = async () => {
      await NfcManager.cancelTechnologyRequest();
    };

[... 227 lines cut ...]
              style={{
                marginTop: 5,
              }}>{`Parsed Text:\n${nfcProps.parsedText}`}</Text>
          )}
          {nfcProps.firstBlockInSector && (
            <Text
              style={{
                marginTop: 5,
              }}>{`First block in sector:\n${nfcProps.firstBlockInSector} [${nfcProps.blocksInSector} blocks]`}</Text>
          )}
        </View>

        <TouchableOpacity
          style={{ marginTop: 20, alignItems: 'center' }}
          onPress={_clearMessages}>
          <Text style={{ color: 'blue' }}>Clear above message</Text>
        </TouchableOpacity>
      </View>
    </ScrollView>
  );
};

export default App;

```

closed 2020-10-15 by BravenxX

javix64, 2023-03-05:
This code should be in DOCS, it's a treasure. Thanks Ghost!