revtel/react-native-nfc-manager #338: [iOS] Tag is not discovered

abdelhakimbendjabeur, 2020-10-15:
Hello, 
I have been unsuccessfully trying to make the example work on iOS 14 (iPhone XS). (I have a developer certificate bought just for this haha).

* react-native-nfc-manager-2.2.0
* react-native-cli: 2.0.1
* react-native: 0.63.3

Here is the code
```
// App.js
import React from 'react';
import {View, Text, TouchableOpacity} from 'react-native';
import NfcManager, {NfcEvents} from 'react-native-nfc-manager';

class AppV2 extends React.Component {
  componentDidMount() {
    NfcManager.start();
    NfcManager.setEventListener(NfcEvents.DiscoverTag, (tag) => {
      console.warn('tag', tag);
      NfcManager.setAlertMessageIOS('I got your tag!');
      NfcManager.unregisterTagEvent().catch(() => 0);
    });
  }

  componentWillUnmount() {
    NfcManager.setEventListener(NfcEvents.DiscoverTag, null);
    NfcManager.unregisterTagEvent().catch(() => 0);
  }

  render() {
    return (
      <View style={{padding: 20}}>
        <Text>NFC Demo</Text>
        <TouchableOpacity
          style={{
            padding: 10,
            width: 200,
            margin: 20,
            borderWidth: 1,
            borderColor: 'black',
          }}
          onPress={this._test}>
          <Text>Test</Text>
        </TouchableOpacity>

        <TouchableOpacity
          style={{
            padding: 10,
            width: 200,
            margin: 20,
            borderWidth: 1,
            borderColor: 'black',
          }}
          onPress={this._cancel}>
          <Text>Cancel Test</Text>
        </TouchableOpacity>
      </View>
    );
  }

  _cancel = () => {
    NfcManager.unregisterTagEvent().catch(() => 0);
  };

  _test = async () => {
    try {
      const res = await NfcManager.registerTagEvent();
      console.log('RES', res);
    } catch (ex) {
      console.warn('ex', ex);
      NfcManager.unregisterTagEvent().catch(() => 0);
    }
  };
}

export default AppV2;
```
Added to info.plist
```
  <key>NFCReaderUsageDescription</key>
  <string>YOUR_PRIVACY_DESCRIPTION</string>
```

nfcexample.entitlements
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>com.apple.developer.nfc.readersession.formats</key>
	<array>
		<string>NDEF</string>
		<string>TAG</string>
	</array>
</dict>
</plist>
```

The phone does not respond to the card when it's near, and  the console logs 
```
RES undefined
```
right after I tap on the button `Test`

Do you guys have an idea as to what I might be doing wrong?

traxx10, 2020-10-18:
Same here, am unable to write on ios 14

abdelhakimbendjabeur, 2020-10-23:
I have gotten it to work by using a [NFC Reader and empty cards](https://www.amazon.fr/Lecteur-ACR122U-Technologie-13-56MHz-Compatible/dp/B07CTSQMLJ) ! 
First, I encoded a URL in am an empty card using the [NFC ideas app for MAC](https://apps.apple.com/us/app/nfc-ideas/id1245536234?mt=12)
Then, I launched [this example](https://github.com/whitedogg13/react-native-nfc-manager/blob/master/example/AppV2.js)
Works just fine!

If you ever do this, you can add the following log to the `setEventListener` function to see the URL !
```
    console.log(
        'tag payload',
        String.fromCharCode(...tag.ndefMessage[0].payload),
      );
```

However, 
I still can not read **ISO Type IV** cards using the react-native app !

My office badge can not even be read by the [NFC Reader](https://www.amazon.fr/Lecteur-ACR122U-Technologie-13-56MHz-Compatible/dp/B07CTSQMLJ). So it might be a different technology

closed 2021-05-26 by github-actions[bot]