revtel/react-native-nfc-manager #339: Writing doesn't work on IOS 14.1

traxx10, 2020-10-18:
Hello, i am currently writing and scanning on android but on ios, writing doesn't work.

`
let resp = await NfcManager.requestTechnology(
        [
          NfcTech.Ndef,
          NfcTech.MifareIOS,
          NfcTech.Iso15693IOS,
          NfcTech.IsoDep,
          NfcTech.MifareClassic,
        ],
        {
          alertMessage: 'Ready to Erase bracelet',
        },
      );
     Code never resolves or reaches here

      let bytes = Ndef.encodeMessage(formatTagNdefData);
      setIswriting(true);
      setLoading(true);
      setDetectedTag(true);
      await NfcManager.writeNdefMessage(bytes);
      await NfcManager.setAlertMessageIOS('I got your tag!');
`

How do i fix this?

guoxianzhe, 2020-10-23:
same issues

YusufBesim, 2020-11-04:
same issue

montecarlo0704, 2020-11-19:
same issue. anyone managed to solve this issue?

YusufBesim, 2020-11-21:
> same issue. anyone managed to solve this issue?

No. Developer not replied yet

traxx10, 2021-01-17:
Anyone resolved this yet?

avivku90, 2021-02-03:
anyone resolved this ?

fgrs, 2021-05-18:
same issue on ios 14.4.2

fgrs, 2021-05-18:
@whitedogg13 do you have any solution to this?

whitedogg13, 2021-05-20:
@fgrs write NDEF using iOS should work correctly, if you're using a formatted and writable NFC tag.

Please use latest v3, and try the following code snippet:
```javascript
import NfcManager, {
  NfcTech,
} from 'react-native-nfc-manager';

async function writeNdef({type, value}) {
    let result = false;

    try {
      await NfcManager.requestTechnology(NfcTech.Ndef, {
        alertMessage: 'Ready to write some NDEF',
      });

      let bytes = null;
      if (type === 'TEXT') {
        bytes = Ndef.encodeMessage([Ndef.textRecord(value)]);
      } else if (type === 'URI') {
        bytes = Ndef.encodeMessage([Ndef.uriRecord(value)]);
      }

      if (bytes) {
        await NfcManager.ndefHandler.writeNdefMessage(bytes);
        result = true;
      }
    } catch (ex) {
      console.warn(ex);
    } finally {
      NfcManager.cancelTechnologyRequest();
    }

    return result;
}
```

The above code snippet is extracted from [react-native-nfc-rewriter](https://github.com/revtel/react-native-nfc-rewriter).

This app has been published to [App Store](https://apps.apple.com/tw/app/nfc-rewriter/id1551243964), you can also try it and see if it can write to your target NFC tags.

Thanks!

fgrs, 2021-05-20:
@whitedogg13 I forgot to reply, I solved the issue by adding into plist **com.apple.developer.nfc.readersession.iso7816.select-identifiers** and then values from this table [https://www.eftlab.com/knowledge-base/211-emv-aid-rid-pix/](url)

I recon this should be mentioned in the docs

closed 2021-08-05 by github-actions[bot]

NathHorrigan, 2021-11-01:
@whitedogg13 The above fixed my issue, This should deffo be in the docs :)